libimobiledevice/libimobiledevice #520: idevicebackup2 is only generating encrypted backups

tmyroadctfig, 2017-06-20:
Hi,

I'm not sure if I'm doing something wrong here, but for my iOS 10 device (10.3.2) I'm only getting encrypted backups from idevicebackup2. And the backup seems to be encrypted with some password other than my icloud/Apple ID password. Is there any way to turn this encryption off when producing the backup?

Thanks, Luke

```
idevicebackup2 --debug backup backup/
Backup directory is "backup/"
Started "com.apple.mobilebackup2" service on port 63896.
Negotiated Protocol Version 2.1
Reading Info.plist from backup.
Starting backup...
Backup will be encrypted.
Requesting backup from device...
Incremental backup mode.
Sending '570a72f7f9526d4deb9f94f5448f0dbd0cc2cffa/Status.plist' (189 Bytes)
```

nikias, 2017-06-20:
The password is not the iCloud password. I mean it can be the same but only if you set it to be that way. It's a separate password you set via ```idevicebackup2 -i encryption on``` or through iTunes. If you don't know it you can't reset it without losing your device data. If you set it up on a mac you might have ticked the checkbox to save it in the keychain.

tmyroadctfig, 2017-06-21:
I would have initially connected my phone to my old Windows machine, so the password isn't in my Mac keychain unfortunately. I wouldn't have set a password explicitly; do you know if iTunes automatically generates a password? I've seen some forum posts where people claim that iTunes used their Windows login password (although that doesn't work in my case either).

nikias, 2017-06-21:
No. It doesn't randomly select a password. You have to explicitly enable encryption and then type a password.

mexmer, 2017-06-21:
@tmyroadctfig when you connect iOS device to iTunes first time (either on PC or mac) it usually does suggest to backup your device and to setup password.

to turn off backup encryption you need to know password you set.

as to how it works

password is used to generate set of encryption keys, that are stored into iOS device and used to encrypt data whenever you making backup. (eg. for creating backup password is never required after it was set, password is only required to restore your backup), so if you forgot your password there is little to nothing you can do. There are programs for bruteforcing password, but in iOS 10 apple really hardened computation of backup password and keys so it's literally impossible to guess key, unless you have some weak password or common password that is in password dictionaries.

moreover, there is no way to reset backup password, if you don't know old password, only way to "reset" encryption is to do device reset, but then you will loose all data.

your only option if you don't know password for backup is to switch to icloud backup, because icloud backup is encrypted by different set of keys, that is stored on your iCloud account.

you might give a try some of those software for bruteforcing password but expect years of computation even for simple passwords like 5 characters (pre iOS 10 it was possible to crack backup password in days on regular gaming computer if it was 5 characters or less)

xdeng, 2017-06-22:
I seem to have encountered such a situation, And finally did not solve

You can refer to

https://support.apple.com/en-us/HT205220

1.You turned on the iTunes encryption option.
2.Whoever set up your device or transferred your data to a new device turned on the iTunes encryption option.
3.Someone set up an installed profile on your device (often for business or education). If so, contact your system administrator.