libimobiledevice/libimobiledevice #525: Exception in backup iphone by AppleMobileBackup.exe

prateek1108, 2017-07-07:
Hello Sir,

i tried to backup my iphone by command prompt while using ApplemobileBackup.exe by command propt

And got the exception with details 
Mobilebackup stop working.

Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	AppleMobileBackup.exe
  Application Version:	17.1584.0.6
  Application Timestamp:	54dd6d0d
  Fault Module Name:	CoreFoundation.dll
  Fault Module Version:	1.750.34.0
  Fault Module Timestamp:	58f06055
  Exception Code:	c0000005
  Exception Offset:	0009f63a
  OS Version:	6.1.7601.2.1.0.256.48
  Locale ID:	1033
  Additional Information 1:	0a9e
  Additional Information 2:	0a9e372d3b4ad19135b953a78882e789
  Additional Information 3:	0a9e
  Additional Information 4:	0a9e372d3b4ad19135b953a78882e789

mexmer, 2017-07-12:
it's not problem of libimobile, it's problem with your itunes installation.
reinstall your itunes, it that doesn't help, contact apple support.

nikias, 2017-07-12:
As @mexmer pointed out, this has nothing to do with libimobiledevice. Closing.

closed 2017-07-12 by nikias