libimobiledevice/libimobiledevice #526: Can get the iPhone Retina display serial number like MLB Serial number using deviceinfo

omkarzunjurke, 2017-07-11:
I want to fetch unique Number of iPhone Retina Display programmatically.Is there any service from which I can fetch Display Hardware serial number using libimobiledevice commands.

Dantee296, 2017-07-29:
lockdownd_error_t ret = LOCKDOWN_E_UNKNOWN_ERROR;
	plist_t value = NULL;

	ret = lockdownd_get_value(client, NULL, "MLBSerialNumber", &value);
	if (ret != LOCKDOWN_E_SUCCESS) {
		return ret;
	}
	plist_get_string_val(value, device_name);

	plist_free(value);
	value = NULL;

	return ret;

mexmer, 2017-07-31:
you need to know domain and value to request any information from iDevice.

if there is software (either mac or pc), that can do it, then information is queryable, if there is none, then most likely is not.

if you know domain and value ... see ideviceinfo help, how to use it.