libimobiledevice/libimobiledevice #529: Suggestion: build all the projects in one solution on Windows

fishjam, 2017-07-20:
I want use libimobiledevice on Windows, so clone the projects from [libimobiledevice-win32](https://github.com/storoj/libimobiledevice-win32), and build it successful with VS2015.

But I found that it will use the compiled libplist and libimobiledevice, if there are any bugs, people can not fix it easily and it's hard to debug it. So I add libimobiledevice-all.sln to build all the projects.  If you can accept this build method, you can merge it to the repositories by yourselves,  or create a libimobiledevice-win32-build repository and I will create a PR to commit it.

Verification procedures:
  - git clone --recurse-submodules https://github.com/fishjam/libimobiledevice-win32-build.git
  - open libimobiledevice-all.sln in Visual Studio 2015
  - after restore nuget packages,  select men: Build => Batch building => Select All => Rebuild
  - then it will build all the 25 projects with Debug/Release and Win32/x64.

Main Changes:
 - 1.add libimobiledevice-all.sln to build all the projects;
 - 2.add using_3rdParty.props to maintain all the third-party packages, avoid hard code in each vcxproj files, so can upgrade easily;
 - 3.modify all the *.vcxproj according libimobiledevice-all.sln and using_3rdParty.props
 - 4.remove unused nuget packages(openssl.v120.*;  libplist and libusbmuxd etc.), so can restore it quickly.
 - 5.change libusbmuxd from StaticLibrary to DynamicLibrary because we need it. If you don't want this, I can revert it.

Notice:
  - 1.Because I changed vcxproj files configuration, so it can not use the original sln files to build, If accept these change, you must change CI configuration.

mexmer, 2017-07-24:
you should send your proposal to libimobiledevice-win32, not here. since that project is maintained by different person.

fishjam, 2017-07-24:
@mexmer , Sorry for my mistake. 
I can not find any issue int [libimobiledevice-win32](https://github.com/libimobiledevice-win32/libimobiledevice), so I thought that all the issue are maintained in [libimobiledevice](https://github.com/libimobiledevice). 
Now I will close this issue.

closed 2017-07-24 by fishjam