libimobiledevice/libimobiledevice #531: How I can get icloud activation status by libimobiledevice ?

amzil, 2017-07-30:
Hello,

I want to know if I can get information about icloud activation status from usb by libimobiledevice tools? 

Ubuntu 16.04

truonggiang0710, 2017-07-31:
@amzil  libimobiledevice responses cloud status when you activate device.

amzil, 2017-08-01:
@truonggiang0710, thank you for your responce , but how I can check if the device have icloud activated or not?
I have already tested "ideviceinfos"  after and before deleting my device from icloud. but I have the same results.

truonggiang0710, 2017-08-08:
@amzil Check activation response and break out to return if your device has icloud.

closed 2017-08-08 by amzil

daniel-bluesea, 2018-02-08:
@truonggiang0710 your suggestion requires a SIM card inserted to the device.
how can we check activation lock status with a wiped/erased device?

btw, ideviceinfo -q com.apple.fmip -k IsAssociated returns always false when device is in setup wizard after wipe/erase

@amzil did you find your answer? :)

amzil, 2018-02-08:
yes, perfect thank you daniel-bluesea.

amzil, 2018-02-08:
@daniel-bluesea how I can then know if the 'Trust This Computer' message alert  is apparent on the device and not confirmed.

daniel-bluesea, 2018-02-08:
@amzil did you manage to activation lock status for device with no sim card? if so please share :)

amzil, 2018-04-18:
hello @daniel-bluesea, yes , please try  ideviceinfo -q com.apple.fmip -k IsAssociated work under ubuntu linux

nikias, 2018-04-18:
Keep in mind this doesn't work if the device has been wiped clean while activation lock is still enabled.

amzil, 2018-04-19:
@nikias tkx for your response, so how i can verify icloud activation status in this case ?

nikias, 2018-04-19:
When you try to activate you will get a prompt for iCloud credentials. And you will need a sim card inserted.

nikias, 2018-04-22:
@daniel-bluesea correct.