libimobiledevice/libimobiledevice #534: Enable LocationServices Remotely via Libimobiledevice

Dantee296, 2017-08-02:
-Suspected Services-

com.apple.locationd 
com.apple.MapsSupport

Property : LocationServicesEnabled  
Values :  1 or 0

i tried to Start Service  " com.apple.locationd " and it not starting it
i tried to set value 
lockdownd_error_t  lerr = lockdownd_set_value(lockdown, "com.apple.locationd", "LocationServicesEnabled", plist_new_real(1));

not working either 
kindly Help me with this
@nikias

mexmer, 2017-08-02:
you have not posted, what error you got, when trying to start service.

mind not all services can be started remotely, or might require provisioning profile (like some services are only available for developers when debuging in xcode, but not for public)

Dantee296, 2017-08-02:
i m gonna try it after mounting developer image
ll let u know what it says.

you have any idea about enabling Location services remotely
Thanks

mexmer, 2017-08-02:
start service doesn't return anything except success or failure, and in case of success also socket for connection.

not sure, what you mean by empty plist then.

as for starting these services itself, no i don't have any knowledge about them, so i don't know, what proper parameters or request are.

xdeng, 2017-08-10:
not support LocationServicesEnabled ???

Dantee296, 2017-08-10:
No , @xdeng do you hv any research on this ?

closed 2020-04-24 by Dantee296