libimobiledevice/libimobiledevice #535: Launch non developer apps

guillermochica, 2017-08-04:
I need to be able to start apps from my computer using the terminal, as with

> ` idevicedebug run bundle-id`

I want to launch installed apps, like youtube, safari, twitter... Not apps that I develop. I can launch them if they are already running in background, after manually starting them. If they are not launched, idevicedebug run does not work. And either way idevicedebug ends with an error like this:

> `Error: failed to get the task for process 45`

What's going wrong? Is it because they are not apps under development? I am not familiar with iOS development since I'm only using this software to automate starting apps for my experiments. Thanks a lot.

qmfrederik, 2017-08-17:
`idevicedebug` uses the debugger to launch applications on iOS devices. To launch an application and attach the debugger to an application, that application needs to have the `get-task-allow` entitlement. That entitlement is not available for applications that ship with iOS or are available from the App Store.

You can launch about any process on an iOS device using the Instruments protocol, but there's no open source implementation of that protocol that I'm aware of.

guillermochica, 2017-08-18:
Hi, thanks for your answer. I finnally was able to launch the app. It turns out that you can launch any app if you enter `idevicedebug run` command 4 times in a row. The first seconds, the app gets frozen and it eventually crash, but, since it remains in background, the following `idevicedebug run` commands can start the app succesfully. It's a bit dirty, but it works for the moment.

allenxyc, 2018-09-06:
Hi @guillermochica , I have the same problem with you, and cannot make it work even enter the command 4 times in a row, can you explain more on that? If the command is `idevicedebug run com.test.myapp`, what kind of your command will be ?

IntergalacticPenguin, 2018-10-09:
Ill give a shot to the 4-times in a row solution, but its really slow and ugly, any news on this topic? I would love to be able to run any app using bundle id via terminal. 🙏

PaunCostin, 2018-11-23:
I'm also interested in this! Thanks

truonggiang0710, 2018-11-26:
@PaunCostin, Do you have any idea for this?

MLB-BIOMETRIC, 2025-10-17:
Any news on this topic?