libimobiledevice/libimobiledevice #537: can't work with IOS11

mingio, 2017-08-14:
ideviceinfo return -8:

15:06:20 lockdown.c:421 lockdownd_query_type(): success with type com.apple.mobile.lockdown
15:06:20 property_list_service.c:128 internal_plist_send(): sending 4924 bytes
15:06:20 service.c:140 service_send(): sending 4 bytes
15:06:20 service.c:140 service_send(): sending 4924 bytes
15:06:20 property_list_service.c:133 internal_plist_send(): sent 4924 bytes
15:06:20 property_list_service.c:134 internal_plist_send(): printing 4924 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceinfo</string>
	<key>PairRecord</key>
	<dict>
		<key>DeviceCertificate</key>
		<data>
		LS0tLS1CRUdJTiBDRVJUSUZJQ0FURS0tLS0tCk1JSUN1akNDQWFLZ0F3SUJB
		Z0lCQURBTkJna3Foa2lHOXcwQkFRVUZBREFBTUI0WERURTNNRGd4TkRBM01E
		SXoKT1ZvWERUSTNNRGd4TWpBM01ESXpPVm93QURDQ0FTSXdEUVlKS29aSWh2
		Y05BUUVCQlFBRGdnRVBBRENDQVFvQwpnZ0VCQUxPaHdyMUdCS2IwYWF6eGox
		eng5SUN4NmtHYnY2dTdGMDVXblZsd0FNR0VndEgvZVhqVWJZdUNUZ3lhCjd4
		cTZ6UDE0Mkg1bXRwY0k2c3c4WHZuQjhiZjBrVEdtazlLTFMwa29sRFhZd1FH
		QUV6UUNZNHdITitqQTNFNUoKdGpRdDFLNkVQY2RmVFVUUHkram1QZXdtYTdP
		dW8rRlJFR1o1dFdLeG9VbC8vc3FqODRYMDQvb2xnbjNQNW84Zwo3UWlmVmRJ
		NkMvRTFXQTVxRTlubXk1ajZUOTg2K3luTk84Y1Y1d1BnL3J5Vjl4bmdqMFk4
		SGRSVkRoMk9xaW5xCjlXZnNzUmJnOGQyMjJHaTk3SDJOTnZoTGcwWGJrd3Nt
		TnRROFl2NDU1Y2JSdUtkSXY3QWlPR3JIUzlKZEdhYWMKSloydFpyTUF5TXB5
		YVIxMUtSeCsyV3ZNcWg4Q0F3RUFBYU0vTUQwd0RBWURWUjBUQVFIL0JBSXdB
		REFkQmdOVgpIUTRFRmdRVVZsU20yMHNSajBKZWNmM21oL2xPMll5eWg4d3dE
		Z1lEVlIwUEFRSC9CQVFEQWdXZ01BMEdDU3FHClNJYjNEUUVCQlFVQUE0SUJB
		UUNzcVMyME5FM2lQM0dscytWQ3o5OW82MS81bkk3VTFQdDk1L00zeVBGSG5F
		V3gKd0dQd1l1WWt3cTVsN1ZMYmxDZ0hhL282V3VvRHc0d3gxQnkvRE9tRlpC
		ODViZHBISWg1NWhHbnpvRXlXRVlCQQpzSmNpVXNMd1ovWS9QQVd3MU5CbFNi
		blcyYUFsUHJ4YnlIekJWb08xWlJmNUsycGZaN2trT2d0MDl5Z0c5TDhNCkll
		L3ltYzNBNUQydEd4NWQrL3ZMcnplbHJCZ0hNZVpwMlBMdWRwdWY2VjBUT0pV
		ZExKZjdJeElyaS83aUNKT2sKcG1QMDEvQnBkU2trY2NVMlY4QXNyOFI2UVNo
		dmU5QTNLWkVBdS9aVUNFSlorN0d4QWd1SWhGM2JkeHFSb1FtSApkdkZQTG5B
		SEZRT0Fjcks1dDJYeVdkV1FzSjZkRjdEUjIwS0xyend3Ci0tLS0tRU5EIENF
		UlRJRklDQVRFLS0tLS0K
		</data>
		<key>HostCertificate</key>
		<data>
		LS0tLS1CRUdJTiBDRVJUSUZJQ0FURS0tLS0tCk1JSUNtekNDQVlPZ0F3SUJB
		Z0lCQURBTkJna3Foa2lHOXcwQkFRVUZBREFBTUI0WERURTNNRGd4TkRBM01E
		SXoKT1ZvWERUSTNNRGd4TWpBM01ESXpPVm93QURDQ0FTSXdEUVlKS29aSWh2
		Y05BUUVCQlFBRGdnRVBBRENDQVFvQwpnZ0VCQU9rcTIralE0YTFxQ0pCN1gz
		elNCT2dISHBmNG9RcUt1SHhlMXBLMXdVZ1B2c0ZMd0VGSWtRQU1CVGFBCldl
		SXp0ZVRwM3gvK0ZSK1hzMjNMTUhHRUJFcVJXQmJ6VGwrc2hTTkhTd1hzY2Ex
		d0YwUitFdXh6OHEzdHZTaTkKM3NEYUllUFNtR0FIcEMvOElvZXowWERNUUpO
		M3lHOVV1YkRnNGlRTG8vb0N4K3lPSlpZNUxEc1R4M2x3T3huTgpXL1M3WHBU
		WmdZN3oxS3Bia0xOMFErQnpydUM2a1plaXl3QlA5ajRTR2tMS3VhRTA5MjlO
		RnF5TlZRd3NLcno0CnV1MVE0dGNXS3RuaWpydVF4YmZJcjdub3h6Q1lGUHlO
		VE5Ha1NPRUVPN040V2plRllHN0laWXNJUkJ3Y1dFME8KVnliR3dkdXhuNVpY
		Sk9CaWR1dGpCakg2a1djQ0F3RUFBYU1nTUI0d0RBWURWUjBUQVFIL0JBSXdB
		REFPQmdOVgpIUThCQWY4RUJBTUNCYUF3RFFZSktvWklodmNOQVFFRkJRQURn
		Z0VCQUtYbGg3Wkk2cVVTSGx6Z0gwZ05tOEs4CnozVUpOMTJYaWxjOFhzcEIz
		Z1lQQ1FVZXJxODU4dzVoWkpjb1ZzL0RHQkhST2NPVmEzWFNHeHRKUnZDdXlv
		b2cKYkhHZC9wWFVEcTdKMk9sOXNvSU9MMEtQcVJtWkM4OVNrV2ZOTjRLV1I0
		VDY3UlIwWG9GWWhXbjZYY1c1akNRRQpIOUk1c2pEZE5nNHN0T3lRNnNJTXdG
		TkljdmVSMWtKVUFyUktLZVhaMTFlM002SW1zUCt4YWlIVyt2dnZXVUFrCi9y
		WVlpUGhwUExBK0RibEtGL3d5YVJXKzJDWkxkVUNEUDZVMGFrQXR5TmNwY29D
		NzFnOGc5R1g1ZTEvaWNsTkcKTlpMNkl0VkJPK1NJaG8yN3N3QnB3RDBraEIv
		UVp2RkZ6Y3dNcC9ndytYOGVHRWZ3YWVLYktVU254M0J3RkxFPQotLS0tLUVO
		RCBDRVJUSUZJQ0FURS0tLS0tCg==
		</data>
		<key>RootCertificate</key>
		<data>
		LS0tLS1CRUdJTiBDRVJUSUZJQ0FURS0tLS0tCk1JSUNqakNDQVhhZ0F3SUJB
		Z0lCQURBTkJna3Foa2lHOXcwQkFRVUZBREFBTUI0WERURTNNRGd4TkRBM01E
		SXoKT1ZvWERUSTNNRGd4TWpBM01ESXpPVm93QURDQ0FTSXdEUVlKS29aSWh2
		Y05BUUVCQlFBRGdnRVBBRENDQVFvQwpnZ0VCQUxuM2s1QzRGRkswQlFuK2Na
		UEZxTGl3em5Sa3ovTVpFd2Q1Q3lqQ0RFak84RVpNZzVqdG9nZktaSmxUCm00
		Qjc0TWpsNVRRWDRJUzd1b3RHN0d0aEdRdkQ5K2NKRm50R2pPOENxTUdmTGhi
		bXNZYitWZVZpMTJFdW9HK0gKbVZiZVpuWWJNdFV3Yzc0dVlSV3FwKzM1QTJV
		MmJTRzlOME5ldGVXamE1OFJVZkE3TTc2N2JIZzJtd05WSGJZRAp3Z0lQNTgr
		RHRRbjhWeGRrc1B0SDRHenhqS0V0WDNxT1F3ZWtMQ09hZ2tzQ0VTZE1aN3JK
		ejJ4SkJNbFhzRStXCkcxY2pMN1kvMXlNM0wvV29LNCtIMkVLYVBsOEpYVnEx
		Z01nTTNFTVliRjg5bnJiSDEzUnZmUHBCcUNwY3NvdjAKb2w2a2FRdDYvc1c0
		UUN3dWlMRUNWaEpLTlpzQ0F3RUFBYU1UTUJFd0R3WURWUjBUQVFIL0JBVXdB
		d0VCL3pBTgpCZ2txaGtpRzl3MEJBUVVGQUFPQ0FRRUFHM2ZmaGQ4SWgvMlNp
		MHhJZFJTWlFVb3dlc0trWE01Nkg3YnR0ZkhmClhTQ0Q0MGt5SDNuTURzNHd0
		elFZTkEvUGg0d25CT0JtUmFHdmp1dnNkY3F2Y0piQ2RoU0NzZkZ3T0k1cXkz
		SWoKeXl2TWg2Z2NLazFpSm5leVhPUDJXb0ZLcm11UmVKNUFna0orT1ZQOGxw
		NUNLZzJ5WDZKK0t4U25tREdLQVluWQpuU3V5cEVTNTFZVUFuZFBIVUo4Mm8r
		UWxUL1RvTGpscm1zdnVXaEl5a3d2Rml3SEYyY0dqUTdDM0R1M3pyYlQ1Cm11
		N2t2Q203V3RnWFJ5d3V4WW9JZUFKY2ZGU2JnSXkvZG9LZmcrSnFCSlI3Syt4
		Mncwc2FEU0ZQTUJMcEloZkQKNTlNZTY2L3RFRWg1MjI1ZXpheElRZ1MwSHFo
		K3EvTWxGT1pxT09KK1oydnVlZz09Ci0tLS0tRU5EIENFUlRJRklDQVRFLS0t
		LS0K
		</data>
		<key>SystemBUID</key>
		<string>512A54B3-3791-FA00-29A0-B257CA27B1A6</string>
		<key>HostID</key>
		<string>72BBD035-A4AE-6709-2556-4CEA609BA3D7</string>
		<key>EscrowBag</key>
		<data>
		sz0o+ZIYYj72eVOCSQsbNJCsnqTow1DC4LQZP7OXtog=
		</data>
		<key>WiFiMACAddress</key>
		<string>f0:f6:1c:27:97:9f</string>
	</dict>
	<key>Request</key>
	<string>ValidatePair</string>
	<key>ProtocolVersion</key>
	<string>2</string>
</dict>
</plist>
15:06:20 idevice.c:399 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -11 (Success)
15:06:20 service.c:163 service_receive_with_timeout(): could not read data
15:06:20 property_list_service.c:193 internal_plist_receive_timeout(): initial read=0
15:06:20 property_list_service.c:195 internal_plist_receive_timeout(): initial read failed!
ERROR: Could not connect to lockdownd, error code -8

nikias, 2017-08-14:
Basically a duplicate of https://github.com/libimobiledevice/libimobiledevice/issues/510.
You need to make sure to use the _lastest_ code from git.

mingio, 2017-08-14:
So sorry for duplicate issue. The lastest code can work with ios 11 beta5.

closed 2017-08-14 by mingio