libimobiledevice/libimobiledevice #539: Ubuntu 16.04 cannot connect to my iPhone.

mmluOsHimEi, 2017-08-17:
Today I installed the libimobiledevice on Ubuntu 16.04 system, and I tried using my iPhone to connect the Ubuntu 16.04.
I  entered the command at the terminal, but did not happen anything.
When I unplugged my iPhone, and the iPhone was immediately reboot.

Anyone have the same situation ? If so, please give me help.

mmluOsHimEi, 2017-08-24:
Who can help me ? Please.

mexmer, 2017-08-24:
you need to compile libimobile from source, ubuntu doesn't have latest version

mmluOsHimEi, 2017-08-28:
First of all, I appreciate your response. I'll try it.

mmluOsHimEi, 2017-08-28:
@mexmer First of all, I appreciate your response. I'll try it.

mmluOsHimEi, 2017-08-28:
@mexmer I'm sorry man, as if I did not make you understand. I mean that my virtual machine did not successfully mount the iPhone device.

mmluOsHimEi, 2017-08-28:
@mexmer  like this:
[image]

mexmer, 2017-08-28:
doesn't matter if it's virtual machine or not, you still need to have latest version of libimobile, which can be only compiled from sourcecode (here on github), since there are no binary packages for it in any linux distribution.

mmluOsHimEi, 2017-08-28:
@mexmer OK ! Get it. Thanks again.

mmluOsHimEi, 2017-08-28:
@mexmer Hi man, I did as you told me.
I did as the following steps:
1. I install git on my new system(ubuntu 16.04.3) on virtualbox.
2.I git clone the libplist,libusbmuxd,libimobiledevice,libusb,usbmuxd source code from Github.
3.I compile the project and make in installed.

But when I did finish all of the work, That error still extsts. Please excuse my poor English.

mexmer, 2017-08-28:
make sure to uninstall libimobile and related packages from your system, before you use compiled version.

mmluOsHimEi, 2017-08-28:
OK😝

mmluOsHimEi, 2017-08-28:
@mexmer OK☺️

mmluOsHimEi, 2017-08-29:
@mexmer I'm so sad about this operations, I failed. The following are my steps about installing libimobile: At first,I downloaded the VirtualBox the latest version  5.1.26 from the official website, and added its extension packages. After that I installed the ubuntu 16.04.4 on virtual. And then I downloaded the libimobile from https://www.libimobiledevice.org/ website, all the lib were the latest version. And I successfully compiled and installed all of them.But the question is **My iPhone still could not be mount to the ubuntu**.

Can you tell me what should I do ? @mexmer  I hope I can get your help. Thank you.

mmluOsHimEi, 2017-08-29:
ubuntu@ubuntu:~$ sudo passwd
[sudo] password for ubuntu: 
Enter new UNIX password: 
Retype new UNIX password: 
passwd: password updated successfully
ubuntu@ubuntu:~$ sudo apt-get update
Hit:1 http://cn.archive.ubuntu.com/ubuntu xenial InRelease
Hit:2 http://cn.archive.ubuntu.com/ubuntu xenial-updates InRelease                                   
Hit:3 http://cn.archive.ubuntu.com/ubuntu xenial-backports InRelease                                 
Hit:4 http://security.ubuntu.com/ubuntu xenial-security InRelease                                    
Reading package lists... Done                        
ubuntu@ubuntu:~$ sudo apt-get dist-upgrade
E: Could not get lock /var/lib/dpkg/lock - open (11: Resource temporarily unavailable)
E: Unable to lock the administration directory (/var/lib/dpkg/), is another process using it?
ubuntu@ubuntu:~$ sudo apt-get dist-upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  gvfs gvfs-backends gvfs-bin gvfs-common gvfs-daemons gvfs-fuse gvfs-libs intel-microcode
  libgail-common libgail18 libgtk2.0-0 libgtk2.0-bin libgtk2.0-common libxfont1 logrotate
  python3-update-manager snapd ubuntu-core-launcher ubuntu-drivers-common update-manager
  update-manager-core update-notifier update-notifier-common
23 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 14.8 MB/14.8 MB of archives.
After this operation, 4,644 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 snapd amd64 2.26.10 [10.1 MB]
Get:2 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 ubuntu-drivers-common amd64 1:0.4.17.3 [49.4 kB]
Get:3 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 update-manager all 1:16.04.9 [543 kB]
Get:4 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 update-manager-core all 1:16.04.9 [5,330 B]
Get:5 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 python3-update-manager all 1:16.04.9 [31.9 kB]
Get:6 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 update-notifier amd64 3.168.5 [47.8 kB]
Get:7 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 update-notifier-common all 3.168.5 [165 kB]
Get:8 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 logrotate amd64 3.8.7-2ubuntu2.16.04.2 [37.7 kB]
Get:9 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 gvfs-bin amd64 1.28.2-1ubuntu1~16.04.2 [55.2 kB]
Get:10 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 gvfs-backends amd64 1.28.2-1ubuntu1~16.04.2 [300 kB]
Get:11 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 gvfs-fuse amd64 1.28.2-1ubuntu1~16.04.2 [14.2 kB]
Get:12 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 gvfs amd64 1.28.2-1ubuntu1~16.04.2 [91.5 kB]
Get:13 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 gvfs-daemons amd64 1.28.2-1ubuntu1~16.04.2 [112 kB]
Get:14 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 gvfs-libs amd64 1.28.2-1ubuntu1~16.04.2 [110 kB]
Get:15 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 gvfs-common all 1.28.2-1ubuntu1~16.04.2 [45.6 kB]
Get:16 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 libgtk2.0-common all 2.24.30-1ubuntu1.16.04.2 [123 kB]
Get:17 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 libgtk2.0-bin amd64 2.24.30-1ubuntu1.16.04.2 [9,834 B]
Get:18 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 libgail-common amd64 2.24.30-1ubuntu1.16.04.2 [112 kB]
Get:19 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 libgail18 amd64 2.24.30-1ubuntu1.16.04.2 [14.2 kB]
Get:20 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 libgtk2.0-0 amd64 2.24.30-1ubuntu1.16.04.2 [1,775 kB]
Get:21 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 libxfont1 amd64 1:1.5.1-1ubuntu0.16.04.2 [95.1 kB]
Get:22 http://cn.archive.ubuntu.com/ubuntu xenial-updates/restricted amd64 intel-microcode amd64 3.20170707.1~ubuntu16.04.0 [985 kB]
Fetched 13.8 MB in 18s (754 kB/s)                                                                       
Preconfiguring packages ...
(Reading database ... 210349 files and directories currently installed.)
Preparing to unpack .../ubuntu-core-launcher_2.26.10_amd64.deb ...
Unpacking ubuntu-core-launcher (2.26.10) over (2.25) ...
Preparing to unpack .../snapd_2.26.10_amd64.deb ...
Warning: Stopping snapd.service, but it can still be activated by:
  snapd.socket
Unpacking snapd (2.26.10) over (2.25) ...
Preparing to unpack .../ubuntu-drivers-common_1%3a0.4.17.3_amd64.deb ...
Unpacking ubuntu-drivers-common (1:0.4.17.3) over (1:0.4.17.2) ...
Preparing to unpack .../update-manager_1%3a16.04.9_all.deb ...
Unpacking update-manager (1:16.04.9) over (1:16.04.7) ...
Preparing to unpack .../update-manager-core_1%3a16.04.9_all.deb ...
Unpacking update-manager-core (1:16.04.9) over (1:16.04.7) ...
Preparing to unpack .../python3-update-manager_1%3a16.04.9_all.deb ...
Unpacking python3-update-manager (1:16.04.9) over (1:16.04.7) ...
Preparing to unpack .../update-notifier_3.168.5_amd64.deb ...
Unpacking update-notifier (3.168.5) over (3.168.4) ...
Preparing to unpack .../update-notifier-common_3.168.5_all.deb ...
[... 2465 lines cut ...]
checking for uint16_t... yes
checking for uint32_t... yes
checking for uint8_t... yes
[... cut ...]
checking whether make supports nested variables... (cached) yes
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating src/Makefile
config.status: creating man/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
config.status: executing libtool commands
ubuntu@ubuntu:~/Documents/ideviceinstaller-1.1.0$ cd ..
ubuntu@ubuntu:~/Documents$

mmluOsHimEi, 2017-08-29:
Parallels Desktop VM  the same as VB VM:
[image]

mmluOsHimEi, 2017-08-29:
When I unplugin the iPhone from the VM and the phone will immediately automatically restart.

mmluOsHimEi, 2017-08-29:
The iOS10.1.1(14B100) can be attached, but >=iOS 10.3.2 can not.

mexmer, 2017-08-29:
helo, in your long copypaste, i don't see, if you have started new usbmuxd, that you have compiled.
if you have usbmuxd installed, remove it, and launch usbmuxd from /usr/local/sbin, you need to run usbmuxd as root

then with /usr/local/bin/ideviceinfo you should be able to get some info about connected phone
you should see connected device, even if it's not paired

you can also use /usr/local/bin/idevice_id

mmluOsHimEi, 2017-08-29:
@mexmer OK!I will have a try tomorrow !

mmluOsHimEi, 2017-08-30:
@mexmer 
Hello,
The way you say it is still not ability to connect iPhone iOS 10.3.3.
The terminal tips: ``` No device found, is it plugged in ?```again and again.
Which library file do you think is the problem of installation ？Thank you very much.

mexmer, 2017-08-30:
is compiled usbmuxd running? ideviceinfo connects to usbmuxd, which provides communication with connected iOS device.

mmluOsHimEi, 2017-08-31:
@mexmer I've done it many times as you say, but all ended in failure. And I have tried other ways, and I cannot succeed connecting my iPhone 5s iOS 10.3.2.

mmluOsHimEi, 2017-09-01:
@mexmer I just want to know which compiler you are using ? Can you tell me about that (complier's name and version)?

sh518, 2017-09-02:
Although the iPhone and Ubuntu can not connect correctly, but I do not understand why the phone will be forced to automatically restart after unplugging the usb-lightning cable.
@nikias  @FunkyM  @mexmer  @mcolyer  @JonathanBeck  @sladen 
Have you ever seen such a problem before ? I'm looking forward to your response.

mexmer, 2017-09-04:
i'm using regular gcc, i'm running it on debian linux tho', not ubuntu. but that should not matter, when you compile from sourcecode.

sh518, 2017-09-05:
@mexmer  Thanks !

closed 2017-09-05 by sh518