libimobiledevice/libimobiledevice #541: AFC_OP_GET_SIZE_OF_PATH_CONTENTS

daniel-bluesea, 2017-08-20:
i think there is no "public" api for getting a path content size, although the above operation is defined in the afc.h file..

i manage to add and use it myself, but i wanted to know if there is a plan to add an official one in the future?

i think its a legitimate feature to know media paths sizes

mexmer, 2017-08-28:
problem is, it doesn't work.
if you send packet with /PhotoData path, for example, it returns object not found error, i tried on lot of folders that are accessible trough AFC service, and except root folder, other folders were returning object not found.

it's possible we just don't know, how to use it. if you know and find it, it can be implemented, but without knowledge, legitimate or not, it's not possible.

daniel-bluesea, 2017-08-30:
i encounter the "strange" behavior that you mentioned regarding "not found" media directories,

i think i manged to get it work by concatenating "itunes" to the path string.

i thought this behavior is because i did something wrong and i hoped that an official development would solve it :)

mexmer, 2017-08-30:
i had success using this on jailbroken phone, with AFC2 service enabled.
but it takes a long time to get response, might be just as slow as doing it yourself from listing folder.

had not found a way how to use it with AFC service (media sandbox), except root folder, every folder returns not found.

daniel-bluesea, 2017-08-30:
@mexmer thanks for your input. 
but i manged to use this AFC_OP_GET_SIZE_OF_PATH_CONTENTS on several non-jailbreak iphones (different models and os) and it works fine for every media subdirectory (with the weird work around i mentioned above)

mexmer, 2017-08-30:
can you post, what needs to be inserted? i will compare it with file structure on jailbroken phone, it's possible, there is similarity in that.

daniel-bluesea, 2017-08-30:
its pretty straight forward, add a new LIBIMOBILEDEVICE_API (afc.c/afc.h) that implement the AFC_OP_GET_SIZE_OF_PATH_CONTENTS operation and call it with path argument that starts with "/itunes" 
(i know that the last part sound weird but..)

mexmer, 2017-08-30:
sounds interesting, will check that. and yes, sounds weird, there is no /itunes folder in real filesystem (at least when i check trough phonebrowser on jailbroken phone)
it's possible that afc service (not afc2) has sandbox tied to itunes application, that's why it needs prefixed path

mexmer, 2017-08-30:
@daniel-bluesea 
my tests show following
if i connect to afc2 service, i don't need to use prefix
if i connect to afc service, i do need prefix "/itunes" to every path, like you mentioned

had not figured out what i need to use for application sandbox (document sharing).

while function itself is interesting, and might be worth implementing, you need to pay attention to some details
1) version of iOS  (function availability is iOS 6+)
2) check if you connecting to AFC2 service (jailbroken device) or AFC (itunes media sandbox)
3) there might be huge delay when enumerating folder with lot of content
4) even if you get size of iTunesControl folder, where music and videos are stored, you still need to diferentiate between music and videos, and that can be only done but manually browsing folders, and calculating size of all mp3/m4a/aac and mp4/qt/mov files

daniel-bluesea, 2017-08-31:
i think this issue is closed, and we'll leave it up to @nikias & @FunkyM whether to add it to the afc functionality.

it is possible, simple and has its usage advantages and also some caveats as we both mentioned.

closed 2017-08-31 by daniel-bluesea