libimobiledevice/libimobiledevice #546: Is it possible to backup 4th gen Apple TV?

tmyroadctfig, 2017-08-28:
I was trying to grab a backup of my Apple TV, but even though it's trusted by the computer it doesn't seem to work. I'm guessing it's not really possible to perform a backup, but can someone confirm that?

Thanks, Luke

```
luke$ ./idevicebackup2 -d backup --full ./backup
Backup directory is "./backup"
Started "com.apple.mobilebackup2" service on port 51331.
Could not perform backup protocol version exchange, error code -1
```

 ```
luke$ ./ideviceinfo 
ActivationState: Activated
ActivationStateAcknowledged: true
BasebandStatus: NoTelephonyCapabilty
...
DeviceClass: AppleTV
DeviceColor: unknown
DeviceName: Apple TV
...
ProductName: Apple TVOS
ProductType: AppleTV5,3
ProductVersion: 10.2.2
ProductionSOC: true
...```

 ```luke$ ./idevicepair validate
SUCCESS: Validated pairing with device 4e60f28d0de2915c59d3cca08d69186fda9a1a7b```

mexmer, 2017-08-28:
try to backup with itunes on mac/win, if that cannot back it up, then idevicebackup will not either.

tmyroadctfig, 2017-08-28:
Cool, that's what I thought. It just comes up in iTunes like this:
[image]

So it doesn't seem like backup is possible. https://support.apple.com/en-us/HT203028

mexmer, 2017-08-28:
hmm, if you in recovery mode, then backup is not possible.
is your apple TV working properly?

codebar33, 2017-08-28:
I can confirm that there is no BackupAgent installed on tvOS, so back up is not possible.