libimobiledevice/libimobiledevice #547: ldevicebackup2 doesn't work on IOS 10.3.3

smart2robot, 2017-08-28:
recently, i update my iphone system to 10.3.3,then try to full backup using idevicebackup2 ,
but it doesn't work, the progress bar stopped and sometimes reply the message "Device is not ready yet. Going to try again in 2 seconds...". anyone who can help me? thank you very much!

codebar33, 2017-08-28:
Have a look to our blog post about this issue: https://imazing.com/blog/ios-10-3-3-device-disconnected-error-during-backup-fix

anohren, 2017-10-17:
I have the same issue, and I use a language with special characters, but:

- I'm on 10.3.1
- I don't use APFS (Although I don't know why the article did not include APFS as a criteria in its list after expressly calling it an APFS bug...)

@codebar33 Do you know of any solutions for that case?