libimobiledevice/libimobiledevice #548: ERROR: Could not connect to lockdownd, error code -3

reidhoch, 2017-08-31:
Currently running into an issue using ideviceinfo, I receive `ERROR: Could not connect to lockdownd, error code -3`, although running `ideviceinfo --simple` is giving me the following
```
ideviceinfo --simple
BoardId: 16
BuildVersion: 15A5368a
ChipID: 28672
DeviceClass: iPod
DeviceColor: #e1e4e3
DeviceName: build’s iPod
DieID: 5141470993031718
HardwareModel: N102AP
HasSiDP: true
PartitionType: GUID_partition_scheme
ProductName: iPhone OS
ProductType: iPod7,1
ProductVersion: 11.0
ProductionSOC: true
ProtocolVersion: 2
TelephonyCapability: false
UniqueChipID: 5141470993031718
UniqueDeviceID: 418bcb77a2dba6f08b42adbccc98fc466e3a3fb8
WiFiAddress: fc:d8:48:44:bf:1f
```

vicmans, 2017-09-06:
When I run `ideviceinfo`
```
>ideviceinfo
ERROR: Could not connect to lockdownd, error code -3
```
do not appear the info to my device, only the error message. I have not test run `ideviceinfo --simple`

fordarnett, 2017-10-23:
It seems to be fixed here https://github.com/libimobiledevice/libimobiledevice/issues/552 but that is not working for me. I have updated everything to the latest but still seeing the same issue

fordarnett, 2017-10-23:
I was able to figure out what happened. It seems like something was still referring to an old version of usbmuxd or ideviceinstaller. I specifically uninstalled usbmuxd, ideviceinstaller, and libimobiledevice. I then did brew install --HEAD libimobiledevice and was able to get it to work.

gu-wei, 2017-11-21:
I installed from --HEAD still has the issue

oosnail, 2017-11-24:
the same。。。。

zhouzhq, 2017-12-13:
Got the same error....

ostriandoni, 2018-02-19:
i got the same error as well even though `ideviceinfo --simple` showing the detail info

karolisg, 2018-08-24:
I am getting same error too
```
ios_webkit_debug_proxy -f chrome-devtools://devtools/bundled/inspector.html --debug
ss.add_server_fd(4)
Listing devices on :9221
ss.add_fd(5)
ss.add_server_fd(6)
ss.remove_server_fd(6)
ss.recv fd=5 len=975
ss.add_server_fd(6)
Could not connect to lockdownd, error code -3. Exiting.
ss.remove_server_fd(6)
Unable to attach 39a9e88db934007015bd8b3e1f12fe4d6aef7ee6 inspector
```
What can be the problem?