libimobiledevice/libimobiledevice #552: Support for iOS 11

b1ueshad0w, 2017-09-06:
Support for iOS11.
Currently all tools will run into error like this (or with a different error code):
`ERROR: Could not connect to lockdownd, error code -3`

mexmer, 2017-09-06:
make sure you using latest sourcecode, it supports iOS 11

b1ueshad0w, 2017-09-06:
I have tried this but not work:
`brew install --HEAD libimobiledevice`
Am I using the latest source code with this command?

mexmer, 2017-09-06:
yes, but you also need latest usbmuxd

b1ueshad0w, 2017-09-06:
It worked. Thanks.

closed 2017-09-06 by b1ueshad0w

kaihendry, 2017-11-04:
Archlinux user here with usbmuxd 1.1.0-2 & libimobiledevice 1.2.0-6. I am able to pair and `sudo ideviceinfo -s` but 

    sudo ifuse /mnt/iphone
    Failed to connect to lockdownd service on the device.
    Try again. If it still fails try rebooting your device.

Doesn't work on IOS 11.0.3. Yes, I tried rebooting my iPhone7+ (iPhone9,4)...

 Any ideas?

nikias, 2017-11-04:
You need a more recent version of libimobiledevice. Since a release is still pending you need to get the latest code from git and compile it yourself.

eli-schwartz, 2017-11-05:
Why is a release still pending? The latest release was in February 2015, and since then support has been added for a major version of iOS, and support has been fixed for recent versions of gnutls. This seems like the kind of thing which definitely warrants a new release, assuming nothing is wrong with the latest git master!

https://bugs.archlinux.org/task/56227 (We could backport patches, I suppose, but this is kind of ugly.)

Raensul, 2017-11-08:
Please, write an instruction how to fix this problem manually. I don't know how can i do that. There is  installed newest version of software from official repositories In my linuxmint 18.2. But my iphone 6s + iOS 11.1 doesn't work with linux...

samrocketman, 2017-11-12:
@Raensul see https://gist.github.com/samrocketman/70dff6ebb18004fc37dc5e33c259a0fc

zebaba, 2018-05-13:
I've compiled and installed from the master branch the most recent versions of libimobiledevice software.  The  problem kaihendry reported above on Nov 4 occurs with IOS 11.3, iPhone 7.1, namely  "idevicepair pair" and "ideviceinfo -s" work, but "ifuse  --debug ~/ipad" does not.   The error message is:

Failed to connect to lockdownd service on the device.
Try again. If it still fails try rebooting your device.

The failure path is:

libimobiledevice.c : src/lockdown.c at line 724
   call to this function 
    userpref_read_pair_record(client_loc->udid, &pair_record);
    returns: NULL for pair_record

libimobiledevice.c : common/userpref.c at line 309:
  call to this function, where udid = 4980ab8a172845a00ad22d7af901439a5ff4e87b
    int res = usbmuxd_read_pair_record(udid, &record_data, &record_size)
  returns:      res is -2 , record_data = NULL , record_size = 0

In libusbmuxd:  src/libusbmuxd.c, line 1216:
   call to:
    ret = usbmuxd_get_result(sfd, tag, &rc, &pl);
   returns
   ret=1 , rc=2

The code immediately following implied rc should be zero not two.

Any ideas?  Thanks in advance.

cbowns, 2018-08-24:
I had to reinstall `libimobiledevice` from HEAD in `brew` to fix this for iOS 12.

```
brew uninstall libimobiledevice
brew install --HEAD libimobiledevice
sudo idevice*
```

and run commands with `sudo` to work around #548