libimobiledevice/libimobiledevice #553: libimobiledevice-dev - libimobiledevice-dev error

einsiol, 2017-09-07:
While trying to install packages for the[ ios-webkit-debug-proxy](https://github.com/google/ios-webkit-debug-proxy), I ran into an error related to the libimobiledevice-dev package.

Here is the full log of the install process:
```
sudo apt-get install autoconf automake libusb-dev libusb-1.0-0-dev libplist-dev libplist++-dev usbmuxd libtool libimobiledevice-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
usbmuxd is already the newest version (1.1.0-2).
usbmuxd set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.
 libimobiledevice-dev : Depends: libglib2.0-dev but it is not going to be installed
                        Depends: libgnutls28-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

```
While trying to solve this issue, I tried installing libgilb2.0-dev manually:

```
sudo apt install libglib2.0-dev
..
The following packages have unmet dependencies.
 libglib2.0-dev : Depends: libglib2.0-0 (= 2.48.0-1ubuntu4) but 2.48.2-0ubuntu1 is to be installed
                  Depends: libglib2.0-bin (= 2.48.0-1ubuntu4)
E: Unable to correct problems, you have held broken packages.

```

I then tried to downgrade the libglib2.0-0 but apt then demands to remove the following packages
`libglib2.0-bin lightdm ubuntu-desktop ubuntu-settings unity unity-tweak-tool`

Now, this would obviously break my system.

I guess this issue has to be fixed but is there a way for me to build this packages, without having to downgrade any packages on my system?

Is this issue related to the libimobiledevice-dev package, or more specifically to libglib2.0-dev?

Here is system information:
```
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.2 LTS
Release:	16.04
Codename:	xenial

apt 1.2.19 (amd64)
```

einsiol, 2017-09-07:
An update on this issue. I was able to get this all working by building `libplist` `libusbmuxd` and then `libimobiledevice` from source, and then was able to build and run `ios-webkit-debug-proxy` successfully, which was my intended goal.

So then what the real problem is, is that the installing libimobiledevice-dev on my machine using the package manager does not work.

mexmer, 2017-09-08:
you cannot and should not use libimobile-dev from package manager, since it's different version than what you build from repository.

einsiol, 2017-09-08:
@mexmer Would that mean that the install instruction from [ ios-webkit-debug-proxy](https://github.com/google/ios-webkit-debug-proxy) is incorrect when they tell users to install `libimobile-dev` using apt?

mexmer, 2017-09-08:
instruction are correct, but you need to take in account, that you are not using binaries from your distribution, you have built binaries from source, therefore you need also use dev stuff (headers, shared libraries) from source, otherwise you will be linking proxy against different version of libimobile, than you are using.

instructions simply don't cover this situation, they expect you to use distribution version, not sourcecode version.