libimobiledevice/libimobiledevice #556: view same devices in List

hujiachun, 2017-09-11:
hujiachun:~ hujiachun$ idevice_id -l
ecab65eca01ae1d42874c26e645a33aee78296b6
ecab65eca01ae1d42874c26e645a33aee78296b6

ketan077, 2017-09-12:
are you using mac host machine ?? device & host machine connected same wifi??

kisa112, 2017-09-22:
Are you currently using Wi-Fi syncing on your iPhone?

dnsBlah, 2017-10-10:
I'm having the same issue, and @ketan077, both devices are on the same network.
my machine by cable and iphone's on WiFi, within the same VLan.

How can we prevent this?

ketan077, 2017-10-10:
just disconnect iPhone device on wifi or go to  /var/db/lockdown/ directory & delate "device_id.plist" (ecab65eca01ae1d42874c26e645a33aee78296b6.plist) file and connect device & check or may be that device already sync with iTunes.

dnsBlah, 2017-10-10:
Hi @ketan007,

This is not working. Well.. at least not if I’m on the same VLan. It will keep showing up.

I always do sudo rm -f /var/db/lockdown/* when my ‘wrapper’ starts up, cus I’m talking about thousands of devices a day for our workers here.

Anyways, I even did unpair on the devices and I still see them.
Maybe the device itself remembers it on client side and together with Bluetooth, WiFi, handsoff, Airdrop or whatever its still showing.

It’s not a real problem for me, but trying to get a massive erase,restore,activate,install profile and applications up and running.

So such an issue can be big.. Is there anyway we can have this only rely on USB ?

dnsBlah, 2017-10-10:
cant edit post -.-
with ‘unpair on device’ I mean idevicepair unpair -u and the udid

For now I will do a workaround and try to get info, if it says that the device is not connected then  will discard that udid

ketan077, 2017-10-11:
if you are using option "sync with this iPhone over wifi"  in itunes then pls unmark it and then check