libimobiledevice/libimobiledevice #561: idevicebackup2 failing with lockdown_check_result(): ERROR: MissingValue

corsac-s, 2017-09-16:
Hi,

idevicebackup2 is failing here. I tried first using packages from Debian sid (so 1.2.0) and I got:

`
corsac@scapa: idevicebackup2 backup -d ../bak
Backup directory is "../bak"
ERROR: Could not start service com.apple.mobilebackup2.
`
I tried rebuilding libimobiledevice from git, with --enable-debug-code, and got the attached log:

[backup.txt](https://github.com/libimobiledevice/libimobiledevice/files/1308422/backup.txt)

the interesting part seems to be:

`
16:46:11 lockdown.c:1268 lockdownd_build_start_service_request(): Adding escrow bag to StartService for com.apple.mobilebackup2
16:46:11 property_list_service.c:128 internal_plist_send(): sending 437 bytes
16:46:11 service.c:140 service_send(): sending 4 bytes
16:46:11 idevice.c:373 idevice_connection_send(): SSL_write 4, sent 4
16:46:11 service.c:140 service_send(): sending 437 bytes
16:46:11 idevice.c:373 idevice_connection_send(): SSL_write 437, sent 437
16:46:11 property_list_service.c:133 internal_plist_send(): sent 437 bytes
16:46:11 property_list_service.c:134 internal_plist_send(): printing 437 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>idevicebackup2</string>
	<key>Request</key>
	<string>StartService</string>
	<key>Service</key>
	<string>com.apple.mobilebackup2</string>
	<key>EscrowBag</key>
	<data>
	XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
	</data>
</dict>
</plist>
16:46:12 idevice.c:430 idevice_connection_receive_timeout(): SSL_read 4, received 4
16:46:12 property_list_service.c:193 internal_plist_receive_timeout(): initial read=4
16:46:12 property_list_service.c:202 internal_plist_receive_timeout(): 350 bytes following
16:46:12 idevice.c:430 idevice_connection_receive_timeout(): SSL_read 350, received 350
16:46:12 property_list_service.c:215 internal_plist_receive_timeout(): received 350 bytes
16:46:12 property_list_service.c:241 internal_plist_receive_timeout(): printing 350 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Error</key>
	<string>MissingValue</string>
	<key>Request</key>
	<string>StartService</string>
	<key>Service</key>
	<string>com.apple.mobilebackup2</string>
</dict>
</plist>
16:46:12 lockdown.c:218 lockdown_check_result(): ERROR: MissingValue
ERROR: Could not start service com.apple.mobilebackup2.
16:46:12 lockdown.c:287 lockdownd_stop_session(): stopping session B37F8342-E3E4-4C84-B44F-7D13A4A04EBB
`
Any pointer appreciated. The device is an iPhone 5s running iOS 10.3.3, I can run ideviceinfo just fine. There's no SIM card in the device but I don't think it'd be related.
`

akong77, 2017-09-23:
Me too.My iphone is 4s and use IOS9.Could fix it?

corsac-s, 2018-04-30:
I can't remember exactly what fixed it, but current git master HEAD is ok on iOS 11.3. A release would be nice but meanwhile Debian sid/buster users can enjoy a recent enough snapshot.