libimobiledevice/libimobiledevice #566: ideviceinfo: Could not connect to lockdownd, error code -8

sankalpanand, 2017-09-22:
I went through the similar errors in this issue list. 
None of them seemed to have an error code 8. 

I have tried- 
* `brew install libimobiledevice --HEAD`
* `idevicepair pair -u <id>`
* `sudo chmod 777 /var/db/lockdown`

None of them worked for me. 

I'm running Xcode 9 against iPhone 6 running iOS 11.

Please suggest what I can do here.

deborah-ufw, 2017-10-18:
This happened to me if I was trying to launch an emulator but had a physical iPhone plugged into my laptop. Unplugging the iPhone resolved it.

dgutie10, 2017-12-13:
I'm experiencing the same issue

bloack, 2018-10-23:
The ULTIMATE weapon: Restart.

ganzf, 2018-12-04:
Hello @bloack, can you be more specific ? Should i reboot the device ?

bloack, 2018-12-07:
> Hello @bloack, can you be more specific ? Should i reboot the device ?

yes，I solved the problem by rebooting my computer. good luck.

eadmaster, 2021-03-27:
For me this was thrown because i didn't paired the device first.

josephgiting, 2022-03-22:
After upgraded to iOS 14.4; restart the device with:
> idevicediagnostics.exe restart

About 1min after rebooted, the iOS device lost the USB connection, i.e.:
> ideviceimagemounter.exe ..\..\14.4\DeveloperDiskImage.dmg

would return:
> ERROR: Could not connect to lockdown, error code -8.

**Disable** the Raise to Wake (Settings -> Display & Brightness) solve the issue.

josephgiting, 2022-04-04:
Previous solution seems work on one of my iPhone 8s (iOS 14.4). However, didn't quite work well with another iPhone 8 (15.4).
After much experiment, I find that uninstall and reinstall the Apple device driver solve this issue on various iPhones which I have.

On PC, go to Device Manager->"Universal Serial Bus devices", uninstall "Apple Mobile Device USB Composite Device":
[image]

reboot the PC, and the new Apple device driver should be in Device Manager->"Universal Serial Bus controllers":
[image]