libimobiledevice/libimobiledevice #570: "idevicediagnostics ioreg AppleARMPMUCharger" fails on iPhone8

kcaalexander, 2017-10-04:
On Iphone8, IOClass "AppleARMPMUCharger" returns an empty plist:

<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>IORegistry</key>
        <dict>
        </dict>
</dict>
</plist>

Apparently, I don't see any errors. By any chance, Apple changed the IOClass?

mseld2, 2017-10-04:
They changed the IOClass.

mseld2, 2017-10-04:
Try IOPMPowerSource. This class has power management stuff. It's been around for a long time.

kcaalexander, 2017-10-05:
Still getting a empty list with IOPMPowerSource

elisardofelix, 2017-12-06:
I found the battery class for the iPhone 8 is **AppleSmartBattery**

linpeifei, 2019-03-20:
> Try IOPMPowerSource. This class has power management stuff. It's been around for a long time.

I want to know the KEY list in order to use "idevicediagnostics ioregentry [KEY]".How do you know one of the KEYs is IOPMPowerSource?

ChrisCheney, 2019-08-30:
How do you determine which classes there are available to query?