libimobiledevice/libimobiledevice #573: Help ! ios 11.0.2 Trust alert dialog doesn't appear!

smart2robot, 2017-10-09:
error log below:

Dec  5 14:38:24 raspberrypi usbmuxd[3292]: [14:38:24.942][1] config_device_record_get_host_id: ERROR couldn't get HostID from pairing record for udid f1c4dce0238bea00bb3829e0db46f593abb518fb

what's wrong?

Orbif, 2017-10-09:
Try to pair your device with this : 

idevicepair -u f1c4dce0238bea00bb3829e0db46f593abb518fb pair

dnsBlah, 2017-10-10:
If you don't have iTunes, or did not open iTunes a first time, or did not accept the terms from iTunes, you won't get the Trust popup, because iTunes is responsible for this pairing normally.

Orbif provided the solution for you :)

JM47459, 2017-10-22:
I tried the above idevicepair -u(UDID) but I still cannot view any files, pics, videos, etc. Has anyone had any luck or am I missing something? 

Much Obliged.