libimobiledevice/libimobiledevice #574: AppleTV: ideviceinstaller doesn't return any result

Parabak, 2017-10-11:
Hi guys, 

I have MAC with macOS version 10.13 and apple tv 4k (new one) iOS11. AppleTV has connected with MAC wirelessly.
Some of the commands are working for me pretty well, for example:
idevice_id -l
idevicename -d -u 8cf34...					
ideviceprovision -u 8cf34... -d list 		
idevicediagnostics restart -u 8cf34...    	
ideviceinfo -u 8cf34...

However ideviceinstaller and idevicedebug are not working. There is no any error, just no response. Looks like command can't make exit. I'm trying to get list of apps and run them:
ideviceinstaller -l -u  8cf34...
idevicedebug run com.**.*** -u 8cf34...

screenshot: https://pasteboard.co/GOqcaCP.png

Installation steps:
 
brew uninstall ideviceinstaller
brew uninstall libimobiledevice
brew install --HEAD --with-ios11 libimobiledevice
brew link --overwrite libimobiledevice
brew install --HEAD ideviceinstaller
brew link --overwrite ideviceinstaller

Best, Alex.

Parabak, 2017-10-13:
Guys, I've updated the info. I've installed libimobiledevice with debug support, also  added the output. Please, help.

Parabak, 2017-10-13:
Maybe the reason is that tvOS is connected via wifi-connection. However the newest apple tv doesn't have an USB-port at all.

`idevicepair -d validate` also failed: 
[idevicepair-stdout.txt](https://github.com/libimobiledevice/libimobiledevice/files/1382931/idevicepair-stdout.txt)
I was trying to dig in and  found one enum:
`enum connection_type {`
	`CONNECTION_USBMUXD = 1`
`};` 

Maybe that's why lib with devices connected wirelessly has problems.
However `idevice_id -l` still shows my apple TV

felyppers, 2018-09-11:
Hey guys, I had this same issue this week and solved by disabling the wifi service in the MAC.

In your MAC, use the command below to list your network services
networksetup -listallnetworkservices

Write down the service name similar to iPhone USB and run:
networksetup -setnetworkserviceenabled SERVICE_NAME off

After that, restart your mobile and it should work