libimobiledevice/libimobiledevice #575: Few commands not working when iPhone connected to Mac via Wifi

StevieSun, 2017-10-12:
I have an iPhone 6 running iOS 9. When i execute any of the commands connected via USB cable, all is fine.

When i enable **"Sync with this device over Wi-Fi"** and disconnect the USB cable, `idevice_id` lists the iPhone connected, `idevicescreenshot` is able to capture screenshot, but `ideviceinstaller` which lists all installed packages is not working and just blocks the terminal.

 Any help regarding this is much appreciated. Thanks!

Parabak, 2017-10-13:
Hi Stevie, I think I have the same problem here. In my case I have an appleTV 4k. Here is my issue:
https://github.com/libimobiledevice/libimobiledevice/issues/574

StevieSun, 2017-10-16:
Thanks Parabak. Hope it gets fixed soon!