libimobiledevice/libimobiledevice #576: Documentation

Koizum1, 2017-10-14:
Is there any documentation on how to use this library? I can find examples, but they have no details or comments of what each line does.

rbelew, 2018-01-09:
1+ !  just running across this great looking library.  but why doesn't have at some doc?

GregoryCollett, 2018-03-21:
+1, seems like a really interesting project. Just a shame it has no supporting documentation

mexmer, 2018-03-21:
you have examples of usage in tools folder.

SilverRampart, 2020-06-25:
https://www.libimobiledevice.org/docs/libimobiledevice/latest/ (See especially the [files](https://www.libimobiledevice.org/docs/libimobiledevice/latest/files.html) tab)
The project uses Doxygen for docs and the docs are reasonably complete. These are generated from comments in the public header files in `include/libimobiledevice` and can also be read there.