libimobiledevice/libimobiledevice #577: MobileSync data class com.apple.Notes not working with iOS 11

RobLinux, 2017-10-16:
When I try to initiate the sync I get "No Data Store available for data class" from the device.
Any clue ?

mexmer, 2017-12-06:
Support for synchronizing notes has been removed.
Apple was silently removing notes synchronization since iOS 9 when they moved to "Modern notes" which supports picture attachments, multiple accounts, organization into subfolders.

Data are also stored in different files.

On iOS 9 and 10 sync was "working" but only syncing text notes stored into "local" account (usually named local (PC) or local(MAC)

on iOS 11 if you go to settings->notes and enable "From My PC Account", you will no longer get error above, but you still will sync 0 notes.

apple is slowly, but persistently forcing people to use their cloud services.

xdeng, 2017-12-13:
new iTunes synchronizing notes has been removed