libimobiledevice/libimobiledevice #582: How to build libimobiledevice on Windows?

smart2robot, 2017-10-24:
i want to use library of libimobiledevice on windows, and implements functions of export and backup ios.
i guess there must have methods to compile the libraries on windows,but how to do it?

ivan, 2017-10-25:
Maybe take a look at https://github.com/libimobiledevice-win32/libimobiledevice (note: I have not tested this)

Mradr, 2017-10-30:
I am still surprised we don't have tools already setup to auto build for windows

deakjahn, 2017-10-31:
@Mradr: They're on NuGet, you just have to include them in your project if you use a compatible IDE.

qmfrederik, 2018-10-03:
We maintain a fork of libimobiledevice which provides binaries for Windows, macOS and Ubuntu. We also provide the NuGet package @deakjahn refers to.

We try to frequently rebase on top of the latest versions of libimobiledevice and keep the delta with 'vanilla' libimobiledevice small.

You can find the latest binaries at https://dev.azure.com/libimobiledevice-win32/imobiledevice-net/_build/results?buildId=264&view=logs. Click Artifacts > Binaries to download the binaries.

oddgames-david, 2019-02-27:
@qmfrederik I've not been able to build that repo, been trying for 3 hours. First you get stuck with unable to load file errors due to the paths not pointing to the package include folders. Then you get an error with DEVICE_LOOKUP_USBMUX being undeclared.

qmfrederik, 2019-02-28:
@oddgames-david We've moved from NuGet to VcPkg for the native (libimobiledevice) dependencies. If you just want to download the binaries, you could go to https://dev.azure.com/libimobiledevice-win32/imobiledevice-net/_build/results?buildId=419 where you can find them in the Artifacts folder.

Alternatively, https://github.com/libimobiledevice-win32/libimobiledevice-vs has a single Visual Studio solution for the core libimobiledevice parts (though ideviceinstaller isn't part of it, yet)

You can also cross-compile from Linux, I think that's how the maintainers of this project do it.

nikias, 2019-02-28:
FYI I compile it with MinGW on Windows.

oddgames-david, 2019-02-28:
@qmfrederik @nikias thanks for the info, i got it compiling with cygwin, been a bit of a learning curve.

wangyuting-art, 2021-03-19:
> @oddgames-david We've moved from NuGet to VcPkg for the native (libimobiledevice) dependencies. If you just want to download the binaries, you could go to [...]

I can't find Artifacts > Binaries in https://dev.azure.com/libimobiledevice-win32/imobiledevice-net/_build/results?buildId=419.  Maybe it has changed?

fsinnes, 2021-07-22:
For the love of god, someone please send me the binaries, ive tried for 3 days straight to build this thing and I cant. 36 hours total. I have tried with the VS solution and with msys.

longeand, 2022-06-12:
I've tried compiling it with MinGW and the Cygwin terminal but am getting errors.  It looks like Windows doesn't support gdb anymore?  I'm trying again be installing a bunch of the dependencies using Cygwin's setup tool....   I did find gdb in Cygwin packages, now i need to figure out the correct paths to use during the config process.  I would love to get libimobiledevice running on Windows.

mexmer, 2022-06-13:
i did compile libimobile on windows with msys2

MasterCna, 2022-07-06:
@mexmer Can please share your steps or make a repo for it ?

mexmer, 2022-07-11:
i had installed msys from https://www.msys2.org/
inside msys i installed following packages
`
gcc
python-devel
openssl-devel
automake
autoconf
`
pacman will automatically take care of dependencies, so i'm listing only main packages
then i did compile packages in following order
libplist
libimobiledevice-glue
libusbmuxd
libimobiledevice

i don't need cython support, so when i got python error after running autogen.sh  (for libplist and libimobile) i just did run
`
configure --without-cython
`
then compiled

yuzhujiutian, 2023-03-06:
> 

configure libplist
`./configure LIBS="-L/usr/lib/python3.11"
`

Why I can get some errors as follows ( I hope no errors ^_^ ):

```
configure: Cython executable is '/usr/bin/cython'
checking for python3.11... (cached) /usr/bin/python
checking for a version of Python >= '2.1.0'... yes
checking for the distutils Python package... yes
checking for Python include path... <string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
<string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
-I/usr/include/python3.11
checking for Python library path... <stdin>:4: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<stdin>:4: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
<stdin>:3: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<stdin>:3: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
<stdin>:2: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<stdin>:2: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
-L/usr/lib -lpython3.11
checking for Python site-packages path... <string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
/usr/lib/python3.11/site-packages
checking python extra libraries... <string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
-lintl -ldl
checking python extra linking flags... <string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead

checking consistency of all components of python development environment... no
configure: error: in '/E/embed_python_miniapp/gitbinbash_adb_python/tobuild/libplist':
configure: error:
  Could not link test program to Python. Maybe the main Python library has been
  installed in some non-standard library path. If so, pass it to configure,
  via the LIBS environment variable.
  Example: ./configure LIBS="-L/usr/non-standard-path/python/lib"
  ============================================================================
   ERROR!
   You probably have to install the development version of the Python package
   for your distribution.  The exact name of this package varies among them.
  ============================================================================

See 'config.log' for more details
```

mexmer, 2023-03-06:
@yuzhujiutian as i mentioned in my previous post, i compile liplist (and also other libimobile projects) without python/cython support. i don't use this project under python at all. probly nikias will give you answer, but my assumption is, that it was simple not updated to compile with never versions of python.

yuzhujiutian, 2023-03-07:
@nikia
Hi, nikia, thanks for sharing mentioning that the compilation was successful. 
Looking forward to sharing a little more information, such as compiled dependent modules, main steps, etc.

mbparker, 2023-04-05:
I used to use the @qmfrederik 's project because it made things soooo much simpler. It seems that project has been abandoned though. It's far out of sync with this repo. The binaries for Windows that were last published don't work for device backups anymore. This left me with no choice but to build it myself - having read that the latest [unpublished] code does work. 

The process to build this project on Windows would be far less painful with even basic documentation. What took 4 hours of trial and error and googling to accomplish, should have taken 30 mins. @mexmer provided enough clues I was able to get it going. However, it's far more involved when you are setting up a brand new environment for the first time. In my case, it's msys2 on a Windows 11 VM running on an Intel Mac.

I'm not sure if the maintainers here just don't like Windows devs, or if it's just a lack of resources willing to sit down and type out the prerequisites and steps, or some other reason. At any rate, the library is a great solution - if you can get it building.

Just my $0.02 - no disrespect to anyone was intended.

mexmer, 2023-04-05:
thanks for thorough guide. i would definetly recommend using msys2 too.

didix21, 2023-04-20:
Are there any plan for moving from MSYS2 to Native Windows build using Visual Studio for example?

Shyamal-Shah, 2023-05-05:
Hi @mbparker, Thanks for the guide, but I am getting the following errors while executing make on libimobiledevice. 
```
make[3]: Entering directory '/home/user/libimobiledevice/3rd_party/libsrp6a-sha512'
  CC       t_conv.lo
  CC       t_math.lo
t_math.c:36:11: fatal error: openssl/opensslv.h: No such file or directory
 # include "openssl/opensslv.h"
           ^~~~~~~~~~~~~~~~~~~~
```
I have added the flags you mentioned, and I have checked the path '/usr/include/openssl' the file exists.

mbparker, 2023-05-12:
@Shyamal-Shah Not sure what your problem is exactly just from that. I can tell you that I wen't to build the 64bit version last night and encountered all sorts of problems. One of which was very similar to yours. The only way that it would build for me was to uninstall MSYS2 and start with a fresh install - adding the prerequisites using a checklist.  Honestly, not sure why it worked fine after that. It seemed to be an identical environment.

It's much more challenging to build this repo on Windows than Linux or even macOS. 

If I can get enough free time, I'll run through the whole exercise again and double check the build steps. Not sure why there isn't a CI process here with binaries available for download.  

If you're in a pinch, I can provide you with the binaries I compiled - but that's not solving the problem long term obviously.

Not much help, sorry.

nikias, 2023-05-12:
Just leaving this here: https://github.com/libimobiledevice/libimobiledevice/actions

mbparker, 2023-05-12:
@nikias Derp - I stand corrected. I should have have looked at the artifacts, not just the build jobs. Thanks.

Shyamal-Shah, 2023-05-12:
Hi @mbparker, thanks solved the errors with the same steps (After a clean install of MSYS2).
I was wondering if we could build the iFuse in the same steps and access the files of the apps with file sharing enabled. Please let me know if you have got any success with that.

mbparker, 2023-05-12:
@Shyamal-Shah I don't use iFuse, but as far as I know it can't work on Windows since FUSE only exists for Linux and macOS. It's been a long time since I looked into that though. You can create a virtual filesystem on Windows, but when I investigated that, it was quite complicated to do.

Shyamal-Shah, 2023-05-16:
Hi @mbparker, thanks to you i am able to use libimobiledevice however, I noticed that ideviceinstaller is not part of this repo and I need to compile it separately. Do you have any steps for that? 
I tried compiling it but no luck there as well. It requires 'libzip-dev' package but I am not able to find it and install it on MSYS. Could you please help in that matter?

nikias, 2023-05-16:
I am using a static libzip to build it, check the workflow here:
https://github.com/libimobiledevice/ideviceinstaller/actions