libimobiledevice/libimobiledevice #584: About idevicenotificationproxy

3208287103, 2017-10-30:
I run this command in terminal:` idevicenotificationproxy -h`
And the terminal shows the following messages:
```
Usage: idevicenotificationproxy [OPTIONS] COMMAND
Post or observe notifications on a device.

 Where COMMAND is one of:
  post ID [...]		post notification IDs to device and exit
  observe ID [...]	observe notification IDs in the foreground until CTRL+C or signal is received

 The following OPTIONS are accepted:
  -d, --debug		enable communication debugging
  -u, --udid UDID	target specific device by its 40-digit device UDID
  -h, --help		prints usage information

Homepage: <http://libimobiledevice.org>
```
According to this description, I still can not use this tool.
Does anyone know how to use this tool ? Please help me. Thanks very much.

@nikias @FunkyM

nikias, 2017-10-30:
The output you posted is explaining exactly how to use it...

closed 2017-10-30 by nikias

3208287103, 2017-10-30:
Thanks you for your reply, Is it supports both remote and local notifications ?

nikias, 2017-10-30:
Neither. It's for system notifications like "com.apple.mobile.application_installed". You can find some examples here:
https://github.com/libimobiledevice/libimobiledevice/blob/master/include/libimobiledevice/notification_proxy.h

3208287103, 2017-10-30:
👌