libimobiledevice/libimobiledevice #586: idevicediagnostics won't restart or shutdown device unless jailbroken

leftyfl1p, 2017-11-13:
I've been using mostly the restart command for the past few days quite frequently and for whatever reason it has stopped working. I still get the expected `Restarting device.` output and all of the other tools in libimobiledevice work fine. The sleep command in idevicediagnostics also works still, only restart and shutdown don't perform anything on the device. Targeting the device by UDID doesn't change anything, as well as using the -d flag. On my 6S 10.0.2 as soon as I run yalu, the commands work as expected every time. I have the same issue on an SE on 10.2.1 (which is not jailbroken) which I have also been using the restart command on quite frequently. After successfully running the command on some other devices for the first time I believe it has something to do with the devices themselves. I've tried re-pairing the devices. Tested host machines are OSX 10.12.16 and raspbian stretch. Let me know what other information you need from me.

vinodvpatil, 2018-10-22:
I am also facing the same issue. @leftyfl1p you got this problem fixed ?

kwongtailau, 2019-04-29:
I am also facing the same issue on iPhone se on 10.3.2. @leftyfl1p

nikias, 2019-04-29:
You need to delete the file /sbin/reboot which is not present on stock iOS.