libimobiledevice/libimobiledevice #587: Could not connect to lockdownd, error code -21

23erter, 2017-11-16:
I've done everything step by step but when I write in CMD "ideviceinfo" it tells me: "Could not connect to lockdownd, error code -21"

What should i do to solve it???

gu-wei, 2017-11-17:
I have the same issue

raulsiles, 2017-11-21:
The "ERROR: Could not connect to lockdownd, error code -21" (for example, simply running "ideviceinfo") is due to a permissions problem in macOS. By default the "/var/db/lockdown" directory has "700" permissions and it is owned by "_usbmuxd:_usbmuxd", like in:
'''
$ sudo ls -ld /var/db/lockdown
drwx------  2 _usbmuxd  _usbmuxd  544 Nov 1 14:45 /var/db/lockdown
'''

Changing the permissions for everyone solves the "-21" error:
'''
$ sudo chmod 705 /var/db/lockdown 
$ sudo ls -ld /var/db/lockdown
drwx---r-x  2 _usbmuxd  _usbmuxd  544 Nov 1 14:58 /var/db/lockdown
'''

My question is: What should be the right (or minimum) set of permissions for "/var/db/lockdown" in macOS in order to be able to run all the "idevice*" related commands?

raulsiles, 2017-11-21:
This issue, when "libimobiledevice" has been installed from brew (latest version, v1.2.0) is also mentioned in issue #188 and #356 (and potentially #588). 
As it is due to a permissions issue, it can also be solved using "sudo ideviceinfo" (or other "idevice*" related commands), apart from changing the directory permissions.

CC-q, 2018-12-18:
I have the same issue, and I solved it. It seems that the reason is the version of **usbmuxd** is too old. And I solved it as follows, hope it helps.
brew update
brew uninstall --ignore-dependencies libimobiledevice
brew uninstall --ignore-dependencies usbmuxd
brew install --HEAD usbmuxd
brew unlink usbmuxd & brew link usbmuxd
brew install --HEAD libimobiledevice
brew link --overwrite libimobiledevice
brew install ideviceinstaller
brew link --overwrite ideviceinstaller

raulsiles, 2018-12-19:
A similar procedure, also valid to fix error code -3, is available at the end of issue #717 :
https://github.com/libimobiledevice/libimobiledevice/issues/717

kwongtailau, 2019-04-29:
I delete the old xcode and install the lastest one. It works for me.

matthewhembree, 2021-08-01:
If you stumble upon this in 2021, reference this [comment](https://github.com/libimobiledevice/libimobiledevice/issues/922#issuecomment-642245254). I am trying to use [MVT](https://github.com/mvt-project/mvt/).

TL;DR: `idevicepair unpair` then `idevicepair pair`. No need to change the permissions on `/var/db/lockdown`.

nikias, 2021-08-01:
Thanks @matthewhembree for pointing this out. I edited the above comments with some clarifications.

gelosecurity, 2022-05-18:
> If you stumble upon this in 2021, reference this [comment](https://github.com/libimobiledevice/libimobiledevice/issues/922#issuecomment-642245254). I am trying to use [...]

This worked for me :)