libimobiledevice/libimobiledevice #588: Lockdown Connection Error : LOCKDOWN_E_INVALID_HOST_ID

rdg10, 2017-11-18:
Lockdown is having issues with connecting in all projects.  I'm focusing on the idevicediagnostics right now for battery and other system info.  I have the lastest code for libimobile devices and all accompanying projects.  Any help would be greatly appreciated.

lockdownd_client_new_with_handshake : LOCKDOWN_E_INVALID_HOST_ID
  ->lockdownd_start_session : LOCKDOWN_E_INVALID_HOST_ID

mexmer, 2017-12-06:
you need to do pair device with your computer first

use idevicepair

rdg10, 2017-12-07:
Thank you for that.  Now that I've paired I'm getting a new error...
idevice_activation_request_new_from_lockdownd: Unable to get ActivationInfo from lockdownd

Phi1758, 2018-10-28:
Same.  $idevicepair list , shows that the device is paired with the proper device but it will not activate without two errors.  Both lockdownd.

stoefln, 2022-08-03:
Can somebody enlighten me? What's "idevicepair" and how do I get it? Is it available as command line application?

stoefln, 2022-08-03:
Oh, actually just found out the answer myself. This isn't precompiled, so I need to compile it first.
I started out with a jar version of libimobiledevice downloaded from here: https://mavenlibs.com/jar/file/org.robovm/robovm-libimobiledevice

It would be actually great if this was platform-independent. Is there a way to use idevicepair with this jar?