libimobiledevice/libimobiledevice #594: Output chinese text problem

iDalink, 2017-11-25:
I use `idevicesyslog ` command to output iphone log. But I get chinese output like _\M-e\M-.\M^L\M-f\M^H\M^P\M-d\M-8\M^@\M-d\M-8\M-*\M-d\M-;\M-;\M-e\M^J\M-!\M-o\M-<\M^L\M-f\M^H\M^P\M-e\M^J\M_. Why doesnt print correct chinese text.

My iphone device  is iphone6s, ios10.2.1.
My Mac is 10.12.6.

Kanong017, 2017-11-27:
It should be your output...
It seems that in general, iPhone will not output Chinese log, there are Chinese pinyin is also directly replaced

If you want to see it , you can only decode it yourself

应该是自己输出的吧...
好像一般来说,iPhone不会输出中文的log,有中文也是直接用拼音代替了

要想看,就只能自己解码咯

suzhenghao, 2017-12-26:
I have the same problem. I use the terminal which support Chinese encoding also have this problem. How to fix it?

mexmer, 2018-01-10:
idevicesyslog outputs characters as is. it doesn't know encoding of source received (while message itself is utf-8 log data are send as base64 and decoded aftewards), just redirect output to file and then view file in viewer, if it doesn't detect encoding, change encoding to appropriate one.

devicelogs themselve don't have any encoding specified, so it does direct output to your terminal in whatever encoding phone wrote it (usualy UNICODE, but might be also UTF8), if you don't have same encoding as log is stored in, you will se stuff like this.

from what you posted, i suspect log is written as UNICODE but your terminal interpretes output as 8bit text.

sfraize, 2018-01-10:
Just been diagnosing this.  I'm new to mobile-dev, but not to unicode problems.   Wherever this problem lies, we know in at least some cases it's not a terminal encoding problem.  The first line below is what I get out of Console.app on MacOS 10.12.6, and the second line is what I get from idevicesyslog:

default 09:59:01.422838 -0500	Runner	*Unicode-Log_Test: heart: ❤
Jan 10 09:59:01 SMF-iPhone-6s  Runner(Flutter)[1101] <Notice>: *Unicode-Log_Test: heart: \M-b\M^]\M-$

Note that "\M-b\M^]\M-$" are not terminal encoded bytes -- they are the actual 7-bit bytes BACKSLASH / LETTER-M / DASH / LETTER-b / BACKSLASH, etc, coming straight out of idevicesyslog.  They come out that way on the screen or when output is redirected to a file.  So *something* in the pipeline is processing the unicode high-bytes coming from iOS and mangling them into "printable" 7-bit ascii.  I suppose it's theoretically possibly Console.app is re-processing "\M-b\..." text-strings back into unicode, but I doubt it.

In this case, it appears it could either be the Google Flutter SDK, or something in libmobiledevice (which seems very handy, tnx btw -- sure you know but Flutter requires it for their SDK).  I notice the google folks are currently blaming this on libmobiledevice, I think because they (and I) don't see this problem when running with the iOS simulator or with Android.

Can anyone confirm that they've seen high-byte unicode (UTF-8) pass successfully through idevicesyslog so we can begin to track down where this problem really lies? 

Tnx.

mexmer, 2018-01-11:
as i said, redirect output to file, converting to ascii is not done by idevicesyslog, i use similar code in my project for dumping syslog, except all output i get from function call i write to file. there is no problem with unicode while viewing this files, but if i do cat syslogfile, i get ton of garbage characters on my terminal screen. because terminal converts stdout bytes.

look at idevicesyslog code as well as syslog_relay it's simple and straightforward

syslog_relay just connect to service and receives data and put them into callback function (no conversion done here)
idevicesyslog creates callback for relay and outputs data byte by byte using putchar function .... and here you might say lies problem, because putchar is byte by byte, so if you send to terminal each byte as 8bit char, you can tell what happens. 

i have not tested it (because i have own code, that does writing directly to file), but it might work for you, if you redirect stdout to file eg.
'idevicesyslog >> out.bin' then view that .... or just modivy idevicesyslog, to output to file, instead of to console.

cbracken, 2018-01-12:
A bit of poking and prodding suggests Apple is encoding logged bytes whose high-order bit set using a relatively simple encoding long before the output gets to libimobiledevice. idevicesyslog's output is consistent with Apple Configurator `cfgutil`.

If it's useful to anyone, I've banged up a simple [decoder](https://gist.github.com/cbracken/d88a84370fdde9cbcfd810d944c8f540) in Dart for use in Flutter. Implementing in C or another language with more direct access to the bytes underlying a string should be even less work.

Stringer86, 2019-01-23:
I am having this issue as well in regards to German umlauts.  For example, "Asylanträgen" is shown in xcode logs accuately. However, the idevicesyslog output is "Asylantr\\M-C\\M-$". @cbracken any recommendations?

cbracken, 2019-01-23:
@Stringer86 looks like the Dart-based decoder I'd hacked up went stale with the migration to Dart 2. I updated it just now and tested it on your input string and it looks like it works. I could bang up a C++ or Obj-C version, but I expect I won't have time to get around to it until the weekend.

wanggh8, 2022-05-21:
@cbracken  thanks. Hope to provide a C++ or Obj-C language version

cbracken, 2022-05-21:
It’s been a few years since I wrote this but turns out this encoding is [`vis`](https://developer.apple.com/library/archive/documentation/System/Conceptual/ManPages_iPhoneOS/man3/vis.3.html) which can be decided with [`unvis`](https://developer.apple.com/library/archive/documentation/System/Conceptual/ManPages_iPhoneOS/man3/unvis.3.html#//apple_ref/doc/man/3/unvis).

NyaMisty, 2022-08-29:
So can we make it integrated into idevicesyslog?

It's a trivial thing to implement, and I really have no idea why @nikias cannot add it.

If that's because you don't have spare time, I can make a PR for this.

If additional BSD library is not acceptable, we can simply copy https://github.com/freedesktop/libbsd/blob/master/src/unvis.c and directly compile it. BSD's license is also fully compatible with GPL, so that totally legal. 

What do you think?