libimobiledevice/libimobiledevice #597: Problem Decoding RAW-PANEL-ID

elisardofelix, 2017-12-05:
Hi,

I'm having problems trying to read the data on "raw-panel-id" from the class "AppleARMMIPIDSIDevice" in the ioreg data from the phone.  The data is encoded in base64 but when I try to decode the data then give me no readable data like the following:

**YG7H948A**

But the physic QR code in the LCD gives me the following id:

**G7H948AN**

And the original base64 code is the following:

**o8RZGgZHN0gTOTQ4QQAAAA==**

Can you please help me to know how to decode the data correctly?

codebar33, 2017-12-05:
o8RZGgZHN0gTOTQ4QQAAAA== decoded to ASCII is **��Y��G7H�948A���**, so you can isolate: **G7H948A**
To do so remove the 5 first bytes, then remove the 9th byte and add a N at the end...  Of course you should check other raw-panel-ids too see if that always works. 

Bytes are:
`0xA3 0xC4 0x59 0x1A 0x06 0x47 0x37 0x48 0x13 0x39 0x34 0x38 0x41 0x00 0x00 0x00`
The 3 last ones are 0 so they could be reserved for longer strings so I think the first zero can be considered as a null terminating char.

closed 2019-07-31 by elisardofelix

msk521, 2021-12-15:
Can you tell me how you solved it
@elisardofelix