libimobiledevice/libimobiledevice #598: Device doesn't keep Trust this Computer

airbud, 2017-12-05:
Hello,

I'm using libimobiledevice and whenever I plug in my device, the device asks me to "Trust this computer" every time I plug it in. I thought the device and computer were supposed to remember that the computer was trusted. Do other people have this same experience? I would like my iPhone to remember that I trust my computer and trust it automatically every time I plug it in. 

Maybe libimobiledevice is not saving some pairing code or key that it should be? How can I make my iPhone remember that it has trusted my computer using libimobiledevice?

Thank you

Bud

miminashi, 2017-12-08:
me too.
Every time I connected iPhone to my computer, the phone asks me to "Trust this computer".

my environment:
- iPhone SE iOS 10.3.3
- Linux 3.4.113 armv7l, Ubuntu 16.04.3 LTS
- libimobiledevice 1.2.0

airbud, 2017-12-08:
It looks like the lockdownd_pair function takes client and pair_record.

`lockdownd_pair(lockdownd_client_t client, lockdownd_pair_record_t pair_record);`

However, when idevicepair calls lockdownd_pair function it passes a NULL pair_record. It seems like trust cannot work like this.

```
lerr = lockdownd_pair(client, NULL);

if (lerr == LOCKDOWN_E_SUCCESS) {

printf("SUCCESS: Paired with device %s\n", udid);

}
```

airbud, 2017-12-08:
After a bunch of reading, I learned that pairing is actually done with usbmuxd. Usbmuxd was complaining that the plist wasn't readable. The plist had fine permissions but the lockdown directory in var did not allow the file to be read. Once i fixed the directory permissions it doesn't ask me to trust every time.

closed 2017-12-08 by airbud

derekhuangxu, 2020-05-18:
> After a bunch of reading, I learned that pairing is actually done with usbmuxd. Usbmuxd was complaining that the pair record plist wasn't readable. The plist had fine permissions but the lockdown directory in var did not allow the file to be read. Once i fixed the directory permissions it doesn't ask me to trust every time.

Hi,May I ask how could you fixed the directory permissions?