libimobiledevice/libimobiledevice #600: idevicepair pair doesn't see device, but it appears in lsusb

EvgenyIv96, 2017-12-15:
Hi.

My computer: evg@evg-Parallels-Virtual-Platform:~$ uname -a
Linux evg-Parallels-Virtual-Platform 4.10.0-42-generic #46~16.04.1-Ubuntu SMP Mon Dec 4 15:57:59 UTC 2017 i686 i686 i686 GNU/Linux
Device: iPad Air (1st generation), iOS 10.3.3 no jailbroken

I've successfully installed all libs from this document https://gist.github.com/samrocketman/70dff6ebb18004fc37dc5e33c259a0fc . And device isn't visible when I'm trying to pair it using idevicepair pair command, but it appears in all connected device list. 

evg@evg-Parallels-Virtual-Platform:~$ lsusb
Bus 001 Device 006: ID 05ac:12ab Apple, Inc. iPad 4/Mini1
Bus 001 Device 002: ID 203a:fff9  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
evg@evg-Parallels-Virtual-Platform:~$ idevicepair pair
No device found, is it plugged in?

I've tried on Lubuntu (16.0.4 LTS) both 32bit and 64 bit systems and on Ubuntu (16.0.4 LTS and 17.10) both 32 and 64 bit systems. Also on Ubuntu my ipad reboots when I disconnecting it from computer.

EvgenyIv96, 2017-12-15:
video with reboot:
https://youtu.be/8expDYVAW4o

purplegamba, 2019-02-09:
I'm having the same issue with an iPod.  Did you ever resolve this?

mexmer, 2019-02-10:
only iPod touch (all generatios) runs iOS, other ipods do run custom firmware that doesn't communicate trough libimobile.

FYI ipod touch looks like iphone.

mexmer, 2019-02-10:
@EvgenyIv96 reboots sounds like either fault usb cable or ipad usb connector. there is no reboot command comming from either libimobile or usbmux, unless you send it.