libimobiledevice/libimobiledevice #607: info command not working

c0d3z3r0, 2018-01-12:
$ idevicebackup2 -i backup ~/Stuff/iOSBackup
[...]
Sending 'cf46bfda3fbcc0ebe19e4d2cb0dd52388aa87ae1/Status.plist' (189 Bytes)
Sending 'cf46bfda3fbcc0ebe19e4d2cb0dd52388aa87ae1/Manifest.plist' (75.0 KB)
Sending 'cf46bfda3fbcc0ebe19e4d2cb0dd52388aa87ae1/Manifest.db' (12.8 MB)
Received 46 files from device.
Backup Successful.
$ idevicebackup2 -i info ~/Stuff/iOSBackup
ErrorCode 207: Invalid password when restoring encrypted backup (MBErrorDomain/207). Underlying error: MKBKeyBagUnlock error (-3) (MBErrorDomain/207).

mexmer, 2018-01-16:
on iOS 11, manifest.db is encrypted if you use encrypted backup, so you need to send password to device, when querying info about backup, otherwise it is not able to decode manifest database. but seems sourcecode of idevicebackup2 was not updated to reflect this change.

@c0d3z3r0 please change your subject to  "idevicebackup2 info command not working on iOS 11.1 encrypted backup", since it does work for other backups.

c0d3z3r0, 2018-01-16:
hmm... I have "-i" supplied? shouldn't it prompt me for the password then?

mexmer, 2018-01-16:
@c0d3z3r0 nope, it's not implemented. previous versions of iOS did not encrypt manifest database, only data itself, so they did not require password, therefore password prompt is not implemented at all.

as of iOS 11 (maybe earlier, like 10.1 or 10.2), manifest.db itself is encrypted, you can check that easily, if file content starts with "SQLite" then it's unencrypted, otherwise it's encrypted. so idevicebackup2 needs to send your backup password as parameter, when calling info, decryption as well as parsing of manifest database is done by iOS device, not by libimobile.

if you don't want to wait for fix and need to get info about backup content, there is handfull of software that can read iOS device backup (although most i know is running on Windows platform). If you only require basic information about backup (when it was done, from what kind of device and such), then parsing manifest.plist should be enough for you.

codebar33, 2018-01-16:
@c0d3z3r0

Have a look to iMazing to view an encrypted backup, it's available for free on Mac and Windows: https://imazing.com/iphone-backup-browser-extractor

Disclaimer: I'm one of the developers :)

mexmer, 2018-01-17:
just to give you update, seems apple derped something again, official tool from itunes cannot read encrypted backup info either for iOS 11, at least on windows, that is. have not tested it on mac yet. might know more tomorow (will have mac with HS at hand)

codebar33, 2018-01-17:
@mexmer **AppleMobileBackup** actions **--info**, **--list** and **--extract** are not working anymore on both Mac and Windows. The only way to parse/extract iOS 10/11 backups is to use third-party tools such as iMazing.

varenc, 2019-10-28:
Just an update on this since I ran into the same issue.  There seems to be no way to provide a password to `idevicebackup2 info`

mexmer, 2019-10-29:
@varenc info is command processed by phone, and it's not working on encrypted backup since ios 11, when apple started encrypting manifest database, as you can see from conversation above. you need to use external tool to decrypt database and list content of backup.