libimobiledevice/libimobiledevice #613: idevicedate not working

jduranc, 2018-01-24:
Hi,

I have been using this tool and its not working as expected, calling to set date functions responds success but device not assign the date given. 

sample:

get date
value: 1511971284
date: 08:01:24 Pacific Standard Time 2017

set date
value: 1516809728
date: 08:02:08 Pacific Standard Time 2018

check date
value: 1511971344
date: 08:02:24 Pacific Standard Time 2017

Tested on iPhone 6, iPhone 6P
iOS 11

any idea whats its happens? any help its apprecieated.

regards,

progressnerd, 2018-04-28:
@nikias This is real. It appears that setting the date does not work on any modern versions, seemingly from 10.3.2 onward.

nikias, 2018-04-28:
Apple decided to only allow setting the time in very specific situations: if the date couldn't be set via GPS/Mobile Network, this would still work. Also then it only works once until again the device decides it has been too long since it got a time from GPS. And it should work on the setup screen, i.e. when the device has been wiped clean or restored, but also I think only if the time wasn't retrieved already.

To make it short: this is not a bug in idevicedate, but expected behavior from the device.

nikias, 2018-04-28:
Also idevicedate tells you about it:

https://github.com/libimobiledevice/libimobiledevice/blob/master/tools/idevicedate.c#L50-L51

closed 2018-04-28 by nikias