libimobiledevice/libimobiledevice #614: security issues of libimobiledevice

imlibra, 2018-01-24:
Hey,

I think DeviceCertificate, RootCertificate, HostCertificate and EscrowBag are extremely sensitive information of an iOS device, it should not be printed any time, even though in debug mode.

Otherwise, it is necessary to store these file on a separate user/group, make normal user processes unable to access it, and add support to store them on various hardware-based cryptography devices like HSMs, TPMs etc in future.

mexmer, 2018-01-28:
you know, that you can read these as ANY local user?
so what exactly is your point?

it doesn't matter if you can see in debug message what is already accessible for everyone on your harddrive.

nikias, 2018-01-28:
These can only be seen locally. If someone manages to gain access to your local computer you should be more worried about other more sensitive information, and, as @mexmer pointed out, can be accessed by normal users anyway. If not with direct file access, you can just ask usbmuxd for a specific pairing record and you will get all the information right away.
Also keep in mind that production builds have all debug output disabled.