libimobiledevice/libimobiledevice #616: Impactor64: LOCKDOWN_E_MUX_ERROR

blshkv, 2018-02-02:
Hi, I'm facing the same issue as in issue #510 and issue #537.
However, it is from a different software (http://www.cydiaimpactor.com/).

Could you please point to the correct patch?

The error log is below:

```
bash ./Impactor 
I:0x7f661c006fc8:XXXXX
usbmuxd_connect: Reading connect result...
usbmuxd_connect: Connect success!
17:42:44 libimobiledevice/src/lockdown.c:675 lockdownd_client_new(): device udid: XXXX

17:42:44 libimobiledevice/src/lockdown.c:405 lockdownd_query_type(): called
17:42:44 libimobiledevice/src/property_list_service.c:128 internal_plist_send(): sending 281 bytes
17:42:44 libimobiledevice/src/service.c:140 service_send(): sending 4 bytes
17:42:44 libimobiledevice/src/service.c:140 service_send(): sending 281 bytes
17:42:44 libimobiledevice/src/property_list_service.c:133 internal_plist_send(): sent 281 bytes
17:42:44 libimobiledevice/src/property_list_service.c:134 internal_plist_send(): printing 281 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>impactor</string>
        <key>Request</key>
        <string>QueryType</string>
</dict>
</plist>
17:42:44 libimobiledevice/src/property_list_service.c:193 internal_plist_receive_timeout(): initial read=4
17:42:44 libimobiledevice/src/property_list_service.c:202 internal_plist_receive_timeout(): 297 bytes following
17:42:44 libimobiledevice/src/property_list_service.c:215 internal_plist_receive_timeout(): received 297 bytes
17:42:44 libimobiledevice/src/property_list_service.c:241 internal_plist_receive_timeout(): printing 297 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Request</key>
        <string>QueryType</string>
        <key>Type</key>
        <string>com.apple.mobile.lockdown</string>
</dict>
</plist>
17:42:44 libimobiledevice/src/lockdown.c:421 lockdownd_query_type(): success with type com.apple.mobile.lockdown
17:42:44 libimobiledevice/src/property_list_service.c:128 internal_plist_send(): sending 4894 bytes
17:42:44 libimobiledevice/src/service.c:140 service_send(): sending 4 bytes
17:42:44 libimobiledevice/src/service.c:140 service_send(): sending 4894 bytes
17:42:44 libimobiledevice/src/property_list_service.c:133 internal_plist_send(): sent 4894 bytes
17:42:44 libimobiledevice/src/property_list_service.c:134 internal_plist_send(): printing 4894 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>impactor</string>
        <key>PairRecord</key>
        <dict>
                <key>DeviceCertificate</key>
                <data>
              xxxxx
                </data>
                <key>HostCertificate</key>
                <data>
                xxxx
                </data>
                <key>RootCertificate</key>
                <data>
xxxx
                </data>
                <key>SystemBUID</key>
                <string>XXX
                <key>HostID</key>
                <string>XXX
                <key>EscrowBag</key>
                <data> XXX
                </data>
                <key>WiFiMACAddress</key>
                <string>XXXX</string>
        </dict>
        <key>Request</key>
        <string>ValidatePair</string>
        <key>ProtocolVersion</key>
        <string>2</string>
</dict>
</plist>
socket_receive_timeout: fd=17 recv returned 0
17:42:44 libimobiledevice/src/idevice.c:400 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -11 (Success)
17:42:44 libimobiledevice/src/service.c:163 service_receive_with_timeout(): could not read data
17:42:44 libimobiledevice/src/property_list_service.c:193 internal_plist_receive_timeout(): initial read=0
17:42:44 libimobiledevice/src/property_list_service.c:195 internal_plist_receive_timeout(): initial read failed!
```

mexmer, 2018-02-05:
you have old version of usbmuxd or libimobile (or both), since it's doing validatepair.
you need to use version compiled from master branch

blshkv, 2018-02-05:
I have the latest available snapshots:
```
app-pda/libusbmuxd-1.1.0_pre20170522
app-pda/usbmuxd-1.1.1_pre20180131
app-pda/libimobiledevice-1.2.1_pre20180117
```

Any other ideas?

mexmer, 2018-02-05:
unfortunately, i don't know from what source those snapshots are built, that's why i'm saying you need to compile them from github source.

blshkv, 2018-02-05:
These snapshots pull source from the relevant github project on that commit date.  I can also give you the commit number if required

mexmer, 2018-02-06:
if binaries are current, then problem might be in impactor, that it does validate_pair manually, it should not do that, in iOS 10 i believe, apple removed ValidatePair and device disconnects if you send it . ValidatePair works for all prior version, but not since iOS 10 up.

blshkv, 2018-02-06:
I have posted this link to Impactor's author. Hopefully, he will join our discussion and fix it.

closed 2018-02-06 by blshkv