libimobiledevice/libimobiledevice #619: Can not build on Mac High Sierra.

pablichjenkov, 2018-02-04:
When executing ./autoconf.sh I got the following output. What I am missing?

./configure: line 15993: syntax error near unexpected token `libusbmuxd,'
./configure: line 15993: `PKG_CHECK_MODULES(libusbmuxd, libusbmuxd >= $LIBUSBMUXD_VERSION)'

abardallis, 2018-03-21:
Same issue here! Any progress?

HHMedic, 2018-06-21:
I has the same quetion, but abardallis's solution doesnot help.

nikias, 2018-06-21:
You need to install pkg-config... is anyone actually reading README / Requirements ?

dandv, 2018-09-27:
@nikias: this dependency is a low-level one when used in higher level tasks like [remote debugging Chrome on iOS](http://jonsadka.com/blog/how-to-debug-a-chrome-specific-bug-on-ios-using-remote-debugging/), so people might never see the README when the follow instructions like `brew install --HEAD libimobiledevice`.

Anyway, thanks for the answer, it worked! (`brew link --overwrite pkg-config`, more specifically, since pkg-config was already installed on my machine).

MRZHL, 2019-04-12:
Thanks for answer  , use brew link --overwrite pkg-config ,sucess

closed 2023-01-27 by pablichjenkov