libimobiledevice/libimobiledevice #628: Various issues using iOS 11.2 device

MidiGlitch, 2018-02-21:
I downloaded the build from a month ago and tested on an iOS 11.2 device.
Ideviceinfo works fine
Ideviceinstaller fails: could not start com.apple.notification_proxy
Idevicesyslog fails: could not start service com.apple.syslog.relay
ideviceimagemouter fails: mount_image returned -3 (connection failed??). Note that -h and -l no longer work. Error: stat: -h: No such file or directory.
idevicedebug fails: could not start com.apple.debugserver

mexmer, 2018-02-21:
Are you running current version of usbmuxd? I mean one from github

MidiGlitch, 2018-02-21:
I omitted to mention that I am using the Win version of Libimobile and downloaded the binaries from https://ci.appveyor.com/project/qmfrederik/imobiledevice-net/build/1.2.0.224/artifacts. Looks like usbmuxd was updated recently on there.

mexmer, 2018-02-22:
you need to direct your request to author of that port then.