libimobiledevice/libimobiledevice #633: iOS 11.3 idevicescreenshooter fails

JonGabilondoAngulo, 2018-03-08:
The screenshotter service on the phone seems not to be working as usual. Fails to open a connection.
Has anyone found a way to to make it work ?

mexmer, 2018-03-09:
do you have developer image mounted?

JonGabilondoAngulo, 2018-03-11:
More Info...
I updated my iPhone 6 to 11.3. b4. No problem with screenshots.
But iPhone 8Plus and X do show the same problem with screenshots.

Apple knows something .. Xcode release notes:
**_Xcode may fail to capture a screen shot from an attached device. (36632000)_**

There is some problem or change in the screenshooter service.

mexmer, 2018-03-11:
Thanks for info,  but that suggest problem lies within ios and not libimobile.

JonGabilondoAngulo, 2018-03-13:
Fixed in beta 5.