libimobiledevice/libimobiledevice #635: Installation fails if Xcode path contains spaces in name

everythingisbugs, 2018-03-09:
We have a script that installs libimobiledevice automatically. However, when deploying to other machines, we have commonly run into an issue where the installer fails because Xcode's path contains spaces in its name.

This is the script part which installs libi:
```
echo "Installing libimobiledevice..."
brew install libimobiledevice --HEAD
echo "Done!"
```

The output is this:
```
Installing libimobiledevice...
==> Cloning https://git.libimobiledevice.org/libimobiledevice.git
Updating /Users/user/Library/Caches/Homebrew/libimobiledevice--git
==> Checking out branch master
==> ./autogen.sh
==> ./configure --disable-silent-rules --prefix=/usr/local/Cellar/libimobiledevi
==> make install
Last 15 lines from /Users/user/Library/Logs/Homebrew/libimobiledevice/03.make:
2018-03-09 15:44:06 +0200

make
install

Making install in common
/bin/sh: /Applications/Xcode: No such file or directory
make: *** [install-recursive] Error 1

READ THIS: https://docs.brew.sh/Troubleshooting
```

Performing `xcode-select -p` outputs the following:
```
/Applications/Xcode 9.2.app/Contents/Developer
```
As you can see, the Xcode installation has a space in its name. This is the default behaviour of the installer as we use multiple versions of Xcode so subsequent installs have the version name appended. We have also encountered this issue when the partition was called 'user's hdd', which also resulted in spaces in the filename, and the error was:
```
Making install in common
/bin/sh: /user's: No such file or directory
make: *** [install-recursive] Error 1
```
This indicates that the path is getting cropped by having a space and not being quoted in the installer.

mexmer, 2018-03-09:
did you quote path, when running configure script? if path has not been quoted when entered as parameter for configure script, it will not be quoted as parameter for installer, that is intended behavior.

nikias, 2018-03-09:
I think the problem is not the install path but the use of tools that reside in the Xcode installation directory. Can you find out what it wants to execute? Look inside the generated Makefile to see what it tries to do...

everythingisbugs, 2018-03-09:
I did not run the configure script at all. `brew install libimobiledevice --HEAD` is literally the only step I'm performing. Configure is called automatically through this step. I am using this since libimobiledevice is available through brew (http://brewformulas.org/Libimobiledevice) and it works on every other system with no issue.

Additionally, because I am installing it through brew, the make file is created/failed/deleted immediately and I cannot interrupt the script in time. 

I will attempt to compile-run with the steps provided in the readme and check out the make file. However, it looks like it genuinely is an issue with the make file since the install script for brew does not pass the Xcode path at any point, nor is my script: https://github.com/Homebrew/homebrew-core/blob/master/Formula/libimobiledevice.rb

mexmer, 2018-03-10:
I see, i don’t use brew on linux, i compile from source, so i’m running configure script myself.

Most likely it’s some tool path, that is passed as environment variable, that is causing issue. Can you post your shell environment variables after failed build?