libimobiledevice/libimobiledevice #636: ERROR: Could not connect to lockdownd, error code -3

f15radar, 2018-03-14:
I am running 11.2.6 on my iPhone.  I believe I have everything the latest because I installed libimobiledevice and others using the following script I found for the Raspberry Pi:

sudo apt-get update
sudo apt-get -y upgrade
sudo apt-get -y install vim tmux git build-essential libxml2-dev python2.7 python2.7-dev fuse libtool autoconf libusb-1.0-0-dev libfuse-dev

mkdir iphone_libs && cd iphone_libs

git clone https://github.com/libimobiledevice/libplist.git
git clone https://github.com/libimobiledevice/libusbmuxd.git
git clone https://github.com/libimobiledevice/usbmuxd.git
git clone https://github.com/libimobiledevice/libimobiledevice.git
git clone https://github.com/libimobiledevice/ifuse.git

export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig
echo "export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig" | sudo tee -a /etc/bash.bashrc

cd libplist && ./autogen.sh && make && sudo make install && cd ..
cd libusbmuxd && ./autogen.sh && make && sudo make install && cd ..
cd libimobiledevice && ./autogen.sh && make && sudo make install && cd ..
cd usbmuxd && ./autogen.sh && make && sudo make install && cd ..
cd ifuse && ./autogen.sh && make && sudo make install && cd ..

groupadd -g 140 usbmux &>/dev/null
useradd -c 'usbmux user' -u 140 -g usbmux -d / -s /sbin/nologin usbmux &>/dev/null
passwd -l usbmux &>/dev/null

echo /usr/local/lib | sudo tee /etc/ld.so.conf.d/libimobiledevice-libs.conf
sudo ldconfig

But I continue to get ERROR: Could not connect to lockdownd, error code -3
What am I doing wrong?

nikias, 2018-03-15:
Can you run ```ideviceinfo -d``` to see the exact place it fails?

nikias, 2018-03-15:
Also it might be because of the key generation on the RPi is slow. Try this patch:
https://gist.github.com/nikias/bc855639fed21f6db3800e32da82d8b2
Make sure to restart usbmuxd and to reconnect your device.

f15radar, 2018-03-15:
Unfortunately, I can't get any debug out of idevceinfo, even though I compiled it with debug. I will try the patch.

f15radar, 2018-03-15:
No.  The patch did not work either.

f15radar, 2018-03-16:
So I copied the libimobiledevice.so* from /usr/local/lib to /usr/lib/arm-linux-gnueabihf and now everything works.  See my problems with the debug flag to see why I did this.

closed 2018-03-16 by nikias