libimobiledevice/libimobiledevice #638: Cant get defug flag to show anything

f15radar, 2018-03-14:
While trying to trouble shoot the lockdown -3 error, I compiled libimobiledevice as follows

./autogen.sh --enable-debug-code

but when I use any of the commands with -d, I get no debug output

suggestions? Thanks.

mexmer, 2018-03-14:
just note, after you have rebuild libimobile and usbmuxd with --enable-debug-code, did you install them?

mind if you have already installed distribution version of usbmuxd and libimobile, you need to uninstall them first.

f15radar, 2018-03-15:
Yes.  I did a make and a make install.  I also did a apt-get remove usbmuxd.  libimobile was not installed.  I just cannot understand why I cant get any debug information.

nikias, 2018-03-15:
What does ```which ideviceinfo``` show? And on the resulting path, check the libraries with ldd:
```ldd `which ideviceinfo` ```

f15radar, 2018-03-15:
root@raspberrypi:~/iphone_libs/libimobiledevice# which ideviceinfo
/usr/local/bin/ideviceinfo
root@raspberrypi:~/iphone_libs/libimobiledevice# ldd `which ideviceinfo`
        /usr/lib/arm-linux-gnueabihf/libarmmem.so (0xb6f6b000)
        libusbmuxd.so.4 => /usr/lib/arm-linux-gnueabihf/libusbmuxd.so.4 (0xb6f3e000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb6f15000)
        libssl.so.1.1 => /usr/lib/arm-linux-gnueabihf/libssl.so.1.1 (0xb6eb5000)
        libcrypto.so.1.1 => /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.1 (0xb6ce4000)
        libplist.so.3 => /usr/lib/arm-linux-gnueabihf/libplist.so.3 (0xb6cc7000)
        libimobiledevice.so.6 => /usr/lib/arm-linux-gnueabihf/libimobiledevice.so.6 (0xb6c9b000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb6b5c000)
        /lib/ld-linux-armhf.so.3 (0xb6f81000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb6b49000)
        libgnutls.so.30 => /usr/lib/arm-linux-gnueabihf/libgnutls.so.30 (0xb69bc000)
        libtasn1.so.6 => /usr/lib/arm-linux-gnueabihf/libtasn1.so.6 (0xb699c000)
        libgcrypt.so.20 => /lib/arm-linux-gnueabihf/libgcrypt.so.20 (0xb68cc000)
        libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0xb68a5000)
        libp11-kit.so.0 => /usr/lib/arm-linux-gnueabihf/libp11-kit.so.0 (0xb6847000)
        libidn.so.11 => /lib/arm-linux-gnueabihf/libidn.so.11 (0xb6806000)
        libnettle.so.6 => /usr/lib/arm-linux-gnueabihf/libnettle.so.6 (0xb67bf000)
        libhogweed.so.4 => /usr/lib/arm-linux-gnueabihf/libhogweed.so.4 (0xb6782000)
        libgmp.so.10 => /usr/lib/arm-linux-gnueabihf/libgmp.so.10 (0xb670f000)
        libgpg-error.so.0 => /lib/arm-linux-gnueabihf/libgpg-error.so.0 (0xb66ef000)
        libffi.so.6 => /usr/lib/arm-linux-gnueabihf/libffi.so.6 (0xb66d7000)
        libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xb66aa000)

mexmer, 2018-03-15:
this
libimobiledevice.so.6 => /usr/lib/arm-linux-gnueabihf/libimobiledevice.so.6 (0xb6c9b000)

it should also point to /user/local where you should have your compiled version of libimobile installed

nikias, 2018-03-15:
Yes. Check the timestamps of /usr/lib/arm-linux-gnueabihf/libimobiledevice.so.6 and a possible /usr/local/lib/libimobiledevice*:
```
ls -l /usr/lib/arm-linux-gnueabihf/libimobiledevice.so*
ls -l /usr/local/lib/libimobiledevice.so*
```

f15radar, 2018-03-16:
root@raspberrypi:~/iphone_libs/libimobiledevice# ls -l /usr/lib/arm-linux-gnueabihf/libimobiledevice.so*
lrwxrwxrwx 1 root root     25 Jan  3  2017 /usr/lib/arm-linux-gnueabihf/libimobiledevice.so.6 -> libimobiledevice.so.6.0.0
-rw-r--r-- 1 root root 113036 Jan  3  2017 /usr/lib/arm-linux-gnueabihf/libimobiledevice.so.6.0.0
root@raspberrypi:~/iphone_libs/libimobiledevice# ls -l /usr/local/lib/libimobiledevice.so*
lrwxrwxrwx 1 root staff     25 Mar 15 04:03 /usr/local/lib/libimobiledevice.so -> libimobiledevice.so.6.0.0
lrwxrwxrwx 1 root staff     25 Mar 15 04:03 /usr/local/lib/libimobiledevice.so.6 -> libimobiledevice.so.6.0.0
-rwxr-xr-x 1 root staff 513384 Mar 15 04:03 /usr/local/lib/libimobiledevice.so.6.0.0

f15radar, 2018-03-16:
So I copied the libimobiledevice.so* from /usr/local/lib to /usr/lib/arm-linux-gnueabihf and now debug works!  Thanks.

This solution also fixes the -3 lockdown error I had.

mexmer, 2018-03-16:
you could just remove libimobile package from your system. it should remove that library in /usr/lib, after that all binaries should use library located in /usr/local/lib

closed 2018-03-16 by nikias