libimobiledevice/libimobiledevice #639: com.apple.calenders not working in ios 11.2.5

shmaylibi, 2018-03-15:
please help me in this case, why local calenders syncronization failed? 
I don't get this error "No Data Store available for data class".
how can i find the problem.
Through icloud com.apple.calenders is working. Calenders sync with iphone. But local sync is not working.
[image]
[image]
[image]
[image]
[image]

mexmer, 2018-03-15:
i might review your code later, but did you read this page?

http://www.libimobiledevice.org/docs/mobilesync.html

shmaylibi, 2018-03-16:
Please see this code 
I m using this code for transfer calendar's events.

 mobilesync_error_t CDataToiphone::SetCalenderData(mobilesync_client_t sync_client)
 {

	Statusinfo objtmpstatus;
	CString	strNameofCont = L"";
	int nitem = 0;
	tableData *tempTable = pFirstCalendar;
	KEYString objtmp;
	CString strtmp = L"";
	//get data from LList
	while (tempTable != NULL)
	{
		CreateCountFile();
		KEYString objtmpTable;
		int nnCCounTab = 0;
		int nLenCocccc = tempTable->listOfKeyValue.GetCount();
		for (int i = 0; i<nLenCocccc; i++)
		{
			objtmpTable = tempTable->listOfKeyValue.GetAt(i);
			if (objtmpTable.strKey.Compare(L"title") == 0)
			{
				strNameofCont = objtmpTable.strValString;
				i = nLenCocccc;
			}
		}
		objtmpstatus.strID = strNameofCont;
		int count = 0,
			i = 0,
			err = 0;
		plist_t data = NULL;
		plist_t msg = NULL;
		plist_t response_t = NULL;
		char * response = NULL;
		char * checkresponse = NULL;
		char * sameresponse = NULL;
		char * sameresponse22 = NULL;
		char * sameresponsealert = NULL;
		char * rkey = NULL;
		char * xml = NULL;
		unsigned int xml_size = 0;
		/* Initial handshake */
		msg = plist_new_array();
		plist_array_append_item(msg, plist_new_string("SDMessageSyncDataClassWithDevice"));
		plist_array_append_item(msg, plist_new_string("com.apple.Calendars"));
		plist_array_append_item(msg, plist_new_string("---"));
		plist_array_append_item(msg, plist_new_string("2013-07-29 22-34-00 00"));
		plist_array_append_item(msg, plist_new_uint(107));
		plist_array_append_item(msg, plist_new_string("___EmptyParameterString___"));
		err = mobilesync_send(sync_client, msg);
		plist_free(msg);
		msg = NULL;
		if (err != 0)
		{
			//printf("mobilesync_send %d, exiting.\n", err);
			return MOBILESYNC_E_SUCCESS;
		}
		plist_free(msg);
		msg = NULL;
		err = mobilesync_receive(sync_client, &msg);
		if (err != 0)
		{
			//printf("mobilesync_receive %d, exiting.\n", err);
			return MOBILESYNC_E_SUCCESS;
		}
		response_t = NULL;
		response = NULL;
		response_t = plist_array_get_item(msg, 0);
[... 299 lines cut ...]
					{
						plist_get_string_val(Item, &response);
						sameresponse = response;
						int vipin = 0;
					}
				}
			}
		}
		xml = NULL;
		plist_free(msg);
		msg = NULL;
		msg = plist_new_array();
		plist_array_append_item(msg, plist_new_string("SDMessageFinishSessionOnDevice"));
		plist_array_append_item(msg, plist_new_string("com.apple.Calendars"));
		xml = NULL;
		err = mobilesync_send(sync_client, msg);
		plist_free(msg);
		msg = NULL;
		err = mobilesync_receive(sync_client, &msg);
		xml = NULL;			
		tempTable = tempTable->pnext;
	}
	DeleteCalendarLists();
	return MOBILESYNC_E_SUCCESS;
}