libimobiledevice/libimobiledevice #655: OSX: possibly undefined macro: AM_PROG_CC_C_O AC_PROG_LIBTOOL ...

swaheed2, 2018-04-09:
``` brew reinstall --HEAD libimobiledevice ```

```
==> Reinstalling libimobiledevice 
==> Cloning https://git.libimobiledevice.org/libimobiledevice.git
Updating /Users/floydf/Library/Caches/Homebrew/libimobiledevice--git
==> Checking out branch master
==> ./autogen.sh
Last 15 lines from /Users/floydf/Library/Logs/Homebrew/libimobiledevice/01.autogen.sh:
src/Makefile.am:6:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
src/Makefile.am:6:   If 'LT_INIT' is in 'configure.ac', make sure
src/Makefile.am:6:   its definition is in aclocal's search path.
configure.ac:6: error: possibly undefined macro: AM_INIT_AUTOMAKE
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:33: error: possibly undefined macro: AM_PROG_CC_C_O
configure.ac:34: error: possibly undefined macro: AC_PROG_LIBTOOL
configure.ac:80: error: possibly undefined macro: AC_MSG_ERROR
configure.ac:84: error: possibly undefined macro: AM_CONDITIONAL
configure.ac:93: error: possibly undefined macro: AM_PATH_PYTHON
configure.ac:94: error: possibly undefined macro: AC_PROG_CYTHON
configure.ac:172: error: possibly undefined macro: AS_COMPILER_FLAGS
./configure: line 2273: syntax error near unexpected token `dist-bzip2'
./configure: line 2273: `AM_INIT_AUTOMAKE(dist-bzip2 no-dist-gzip check-news)'
```
Tried:
```autoreconf --install```

```autoreconf: 'configure.ac' or 'configure.in' is required```

```autoconf```, ```automake```, ```libtool``` are installed

pich4ya, 2018-10-13:
up

mexmer, 2018-10-13:
run 
```
autoreconf -fvi
```
you build environment doesn''t have properly set paths or update all necessary tools, this command should fix it.

swaheed2, 2018-10-25:
@mexmer where do you run this command from?

I get ```autoreconf: 'configure.ac' or 'configure.in' is require```

swaheed2, 2018-10-29:
Actually I think it worked. After installing the required packages reinstalling fixed the errors: 

```brew reinstall libimobiledevice``` 

For some reason in the above example I was installing from the HEAD which threw errors:

```brew reinstall --HEAD libimobiledevice```

closed 2018-10-29 by swaheed2