libimobiledevice/libimobiledevice #659: idevicebackup2: stop trying after a while when device is not available

corsac-s, 2018-04-30:
Hi,
when for some reason the device is not available (for example because the it has been unplugged at a bad time), idevicebackup2 loops on:
`Device is not ready yet. Going to try again in 2 seconds...`
and it keeps going on.
It'd be nice to have it exit with an error code indicating failure.

Blefish, 2018-05-02:
I've also noticed this. In the restore script I think there was a 3 minute timeout, but not in idevicebackup2.

corsac-s, 2018-05-02:
There's a timeout on the [lock](https://github.com/libimobiledevice/libimobiledevice/blob/master/tools/idevicebackup2.c#L1738) but I can't find one on restore.

mexmer, 2018-05-12:
it is possible to restore ios device backup to another ios device backup if following conditions are met
1) target device has enough space for restore
2) target device runs same or newer version of iOS as backup device

while it's not recommended to restore ipad over iphone and vice versa, and it possible to do that also.

as for unback, if i'm not mistaken, apple removed that option in ios 11.

corsac-s, 2018-05-13:
On Sat, 2018-05-12 at 02:57 -0700, Josef Micka wrote:
> as for unback, if i'm not mistaken, apple removed that option in ios 11.

I don't think it's been removed. At least it works at first, but at one point
indeed there's some kind of timeout and the host can't reach the device
anymore. I'm unsure if it's possible to do the unbak step by step.
-- 
Yves-Alexis

nikias, 2018-05-13:
Apple broke the unback command, probably not on purpose, but is is broken. It has to do with some of the backup domains and the backup daemon crashes.

corsac-s, 2018-05-13:
On Sun, 2018-05-13 at 03:01 -0700, Nikias Bassen wrote:
> Apple broke the unback command, probably not on purpose, but is is broken.
> It has to do with some of the backup domains and the backup daemon crashes.

Thanks, so the timeout behaviour is actually new in iOS 11 and linked to that?
-- 
Yves-Alexis