libimobiledevice/libimobiledevice #666: AFC_OP_MAKE_LINK not supported on iOS 11

EthanArbuckle, 2018-05-12:
Hello,

iOS 11 devices are rejecting the `AFC_OP_MAKE_LINK` operation command, failing with error "Unsupported API" (From syslog).

I investigated in `libafc.dylib`, support for symlinks over AFC has been removed. In function `AFCProcessServerPacket()`, when `AFC_OP_MAKE_LINK` is encountered (`0x0000001C`), an error is immediately thrown.

See snippet:
```
case 0x1CLL: (AFC_OP_MAKE_LINK)
    v5 = __AFCSetErrorResult_4(1LL, 0xFFFFFFFFLL, 0xE800400FLL);
    goto LABEL_224;
case 0x1DLL: (AFC_OP_GET_FILE_HASH)
    v257 = 0LL;
```

The only solution to installing an app bundle that includes a symlink is to transfer the compressed ipa and let iOS extract it on-device.

nikias, 2018-05-12:
Yes because several jailbreaks have been abusing symlinks over AFC. However there is no need to change anything in libimobiledevice since the afc daemon on the device returns an error that the client can handle.

EthanArbuckle, 2018-05-12:
What can be done documentation wise to indicate `AFC_OP_MAKE_LINK` isn't expected to work on iOS 11+. Possibly a note above the declaration in afc.h, similar to https://github.com/libimobiledevice/libimobiledevice/blob/00f8e5733f716da8032606566eac7a9e2e49514d/src/afc.h#L96

xdeng, 2018-06-07:
emmmmmmmmmmmmmmmmm