libimobiledevice/libimobiledevice #669: Failed to connect to lockdownd service on the device

zebaba, 2018-05-20:
I am opening this issue to report the same problem as mentioned by kaihendry on Nov 4, 2017 issue #552.  I provided the details below in that closed issue, but did not receive a reply.

I've compiled and installed from the master branch the most recent versions of libimobiledevice software. The problem kaihendry reported above on Nov 4 occurs with IOS 11.3, iPhone 7.1, namely "idevicepair pair" and "ideviceinfo -s" work, but "ifuse --debug ~/ipad" does not. The error message is:

Failed to connect to lockdownd service on the device.
Try again. If it still fails try rebooting your device.

The failure path is:

libimobiledevice.c : src/lockdown.c at line 724
call to this function
userpref_read_pair_record(client_loc->udid, &pair_record);
returns: NULL for pair_record

libimobiledevice.c : common/userpref.c at line 309:
call to this function, where udid = 4980ab8a172845a00ad22d7af901439a5ff4e87b
int res = usbmuxd_read_pair_record(udid, &record_data, &record_size)
returns: res is -2 , record_data = NULL , record_size = 0

In libusbmuxd: src/libusbmuxd.c, line 1216:
call to:
ret = usbmuxd_get_result(sfd, tag, &rc, &pl);
returns
ret=1 , rc=2

The code immediately following implied rc should be zero not two.

Any ideas? Thanks in advance.

veraxlim, 2018-05-26:
Just wanted to say I'm having the same problem as you.

samrocketman, 2018-08-20:
If you’re using my instructions, the environment variables such as LD_LIBRARY_PATH are important to be set.  By rebooting you are likely not setting up your paths correctly after rebooting.