libimobiledevice/libimobiledevice #676: Is it possible to use backup of old Iphone with a new Iphone? (same model)

Sol33t303, 2018-06-20:
I have backed up my friends Iphone so he could buy a new one after breaking his old one (I think something may have happened to the networking?). So he's bought a new Iphone that is the exact same model i'm hoping there would be a way to use the backup on the new phone (or at least collect some of the data from the backup like contacts, pictures, etc).

mexmer, 2018-06-20:
you can restore backup from one iphone to another, there are 2 conditions required
1) target phone need to have enough space for backup
2) target phone must have same or newer iOS version as oldphone

Sol33t303, 2018-06-20:
How do you do it then? I tried restoring the phone with the command `sudo idevicebackup2 restore --system --reboot --copy --settings /home/sol33t303/iphone/d218b7106a13d47571bece410d88fa5475ad07a7`
But then it would complain about not being able to find the Info.plist for the phones UDID (here is the complete output).
`ERROR: Backup directory "/home/sol33t303/iphone/d218b7106a13d47571bece410d88fa5475ad07a7" is invalid. No Info.plist found for UDID 45508f7722574a516f09f28e7404b819404aa5ca.`

mexmer, 2018-06-20:
you need to specify source and destination udid with -u and -s respectively, otherwise it will look for target device backup.

```
idevicebackup2 restore --system --reboot --copy --settings -u 45508f7722574a516f09f28e7404b819404aa5ca -s d218b7106a13d47571bece410d88fa5475ad07a7 /home/sol33t303/iphone/
```

Sol33t303, 2018-06-21:
Ah ok, Thanks for the help :+1:

closed 2018-06-21 by Sol33t303