libimobiledevice/libimobiledevice #678: ipheth_tx_timeout: TX timeout

kaihendry, 2018-06-29:
I'm on an updated Archlinux system running 1.2.0+66+g5a85432-1, using the stable 4.14.52-1-lts kernel. After a short time of connectivity on my upto date iPhoneX, no matter the mobile network (I travel in different countries), it seems to conk out with `ipheth_tx_timeout: TX timeout` messages.

If this is the wrong place to file a bug, please let me know where / how I can provide some useful information please.

```
Jun 28 21:13:25 t480s kernel: ipheth 1-1:4.2: Apple iPhone USB Ethernet device attached
Jun 28 21:13:25 t480s kernel: usbcore: registered new interface driver ipheth
Jun 28 21:13:25 t480s kernel: ipheth 1-1:4.2 enp0s20f0u1c4i2: renamed from eth0
Jun 28 21:23:40 t480s kernel: NETDEV WATCHDOG: enp0s20f0u1c4i2 (ipheth): transmit queue 0 timed out
Jun 28 21:23:40 t480s kernel: Modules linked in: ipheth xt_nat xt_tcpudp veth ipt_MASQUERADE nf_nat_masquerade_ipv4 nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 xt_addrtype iptable_filter xt_conntrack nf_nat nf_conntrack libcrc32c br_netfilter bridge stp
llc overlay snd_hda_codec_hdmi uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev media snd_soc_skl snd_soc_skl_ipc snd_soc_sst_ipc snd_hda_codec_realtek snd_soc_sst_dsp snd_hda_ext_core snd_soc_sst_match snd_hda_codec_generic snd_soc_core snd_compress snd_pcm_dmaengine ac97_bus ccm joydev algif_aead cbc mousedev des_generic cmac sha512_ssse3 sha512_generic md4 algif_hash tpm_crb arc4 iwlmvm iTCO_wdt iTCO_vendor_support wmi_bmof mac80211 intel_rapl x86_pkg_temp_thermal intel_powerclamp
Jun 28 21:23:40 t480s kernel: ipheth 1-1:4.2: ipheth_tx_timeout: TX timeout
Jun 28 21:23:44 t480s kernel: ipheth 1-1:4.2: ipheth_tx_timeout: TX timeout
Jun 28 21:23:50 t480s kernel: ipheth 1-1:4.2: ipheth_tx_timeout: TX timeout
Jun 28 21:23:54 t480s kernel: ipheth 1-1:4.2: ipheth_tx_timeout: TX timeout
Jun 28 21:24:00 t480s kernel: ipheth 1-1:4.2: ipheth_tx_timeout: TX timeout
Jun 28 21:24:04 t480s kernel: ipheth 1-1:4.2: ipheth_tx_timeout: TX timeout
Jun 28 21:24:10 t480s kernel: ipheth 1-1:4.2: ipheth_tx_timeout: TX timeout
Jun 28 21:24:14 t480s kernel: ipheth 1-1:4.2: ipheth_tx_timeout: TX timeout
Jun 28 21:24:20 t480s kernel: ipheth 1-1:4.2: ipheth_tx_timeout: TX timeout
Jun 28 21:24:22 t480s kernel: ipheth 1-1:4.2: ipheth_rcvbulk_callback: urb status: -71
Jun 28 21:24:23 t480s kernel: ipheth 1-1:4.2: Apple iPhone USB Ethernet now disconnected
```

0xBE7A, 2018-07-24:
Seems like it got fixed in 4.15 (https://github.com/torvalds/linux/commit/61c59355e0154a938b28710dfa6c1d8be2ddcefa)

Are you still able to reproduce the issue after updating?

kaihendry, 2018-07-26:
I wonder why it broke in the first place?

I just tested 4.17.9-1. It crashed whilst testing it (got that flashing capslock). The logs don't show whether ipeth was responsible or not on my T480s. Tbh I am little clueless how to move forward here other than test another day since the logs have nothing to go on.

0xBE7A, 2018-07-26:
I think we are observing two issues here: The TX_Timeout one was caused by the introduction of "trust this computer" in ios but got fixed in 2017.

Im also observing freezes on the latest kernel after a couple of minutes of tethering, with a lot of crash-logs prior in the kernel log. Seems like someone already filed a bug-report for this [1] but the last activity over there was in March.

[1] https://bugzilla.redhat.com/show_bug.cgi?id=1550181

ozfaro, 2018-09-18:
Are there any news on this bug. The RedHat issue closed. We encounter thios bug everytime we download large files at high speed (>1MB/s) over the ipheth driver. The complete system freezes. It can be reproduced with iOS 11 different versions as well as iOS12 now.
Anybody who has info / help on this issue?

kaihendry, 2018-09-19:
Hello again! I just did a test on 4.18.8-arch1-1-ARCH with IOS 12.0 iPhone X, tethered via USB on my T480s. Package version: 1.2.0+66+g5a85432-2 of libimobiledevice

I conducted a speedtest: http://www.speedtest.net/result/7647957860

https://s.natalian.org/2018-09-19/ipeth.txt

```
5: enp0s20f0u1c4i2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether b2:19:c6:c5:c8:4b brd ff:ff:ff:ff:ff:ff
    inet 172.20.10.3/28 brd 172.20.10.15 scope global dynamic enp0s20f0u1c4i2
       valid_lft 85258sec preferred_lft 85258sec
    inet6 2001:d08:2001:71d4:b019:c6ff:fec5:c84b/64 scope global mngtmpaddr noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fe80::b019:c6ff:fec5:c84b/64 scope link
       valid_lft forever preferred_lft forever
```
So far so good. I'll try use mobile data till the end of the day and if it's OK, I'll close the bug.

Big thank you to whoever fixed it. Here's hoping it doesn't break again.

closed 2018-09-19 by kaihendry

ozfaro, 2018-09-19:
Hello,

on ubuntu 16.04LTS with main line kernel 4.18.8, iPhone 8Plus the bug still appears frequently. I think it is not fixed. The last commit in the kernel sources for ipheth is 10 month old. We are currently lookong at the driver. In general it looks like that the bug occurs more frequently when there is high traffic at a high speed on the interface.

kaihendry, 2018-09-24:
Now I am hitting `ipheth_rcvbulk_callback: urb status: -71` when I am out and about today. :man_facepalming: